jetty840/Sailfish-MightyBoardFirmware #135: need Update docs\avr-gcc.markdown

Dogm, 2015-08-06:
1) This lines must be corrected to last version ()
> curl -O http://www.netgull.com/gcc/releases/gcc-4.6.3/gcc-core-4.6.3.tar.bz2
> curl -O http://www.netgull.com/gcc/releases/gcc-4.6.3/gcc-g++-4.6.3.tar.bz2
> bunzip2 -c gcc-core-4.6.2.tar.bz2 | tar xf -
> bunzip2 -c gcc-g++-4.6.2.tar.bz2 | tar xf -
> cd gcc-4.6.2

Correct url and folders: 
* http://www.netgull.com/gcc/releases/gcc-5.2.0/gcc-5.2.0.tar.gz

2) This lines must be updated up to
> curl -O http://savannah.spinellicreations.com/avr-libc/avr-libc-1.7.2rc2252.tar.bz2
> bunzip2 -c avr-libc-1.7.2rc2252.tar.bz2 | tar xf -
> cd avr-libc-1.7.2rc2252

Correct url and folders: 
* http://savannah.spinellicreations.com/avr-libc/avr-libc-1.8.1.tar.bz2

Dogm, 2015-08-06:
it handle errors in current state

dcnewman, 2015-08-06:
Use gcc 5.x at your own risk.  It may or may not work.  Back in 2012 we were never been able to get gcc 4.7 or 4.8 to produce correct, working code.  The Marlin folks recently tried gcc 4.8 and have encountered problems.  (That happened because newer versions of the Arduino app have moved from 4.3 to 4.8.)  The problems generally manifest while actually printing.  Consequently, we only build Sailfish with the avr-gcc 4.6.2 or 4.6.3 tool chains.

closed 2015-08-06 by dcnewman

Dogm, 2015-08-06:
ok change to correct links please - no matter version
! current file contains broken links

dcnewman, 2015-08-06:
If you know working links for 4.6.2 or 4.6.3, then I can update them.  Problem is, the links do not remain viable for ever so it becomes a constant battle keeping them up to date.  A better alternative may be to find the tarballs for that chain and park them here at github.  I do not recall if I have the tarballs lying about or not.  Will take several days to find out (as I'm traveling at the moment).

Dogm, 2015-08-06:
did you mean this?
http://www.netgull.com/gcc/releases/gcc-4.6.3/gcc-4.6.3.tar.gz
http://www.netgull.com/gcc/releases/gcc-4.6.3/gcc-core-4.6.3.tar.gz
http://www.netgull.com/gcc/releases/gcc-4.6.3/gcc-g++-4.6.3.tar.gz

this links is work for me,
and need correct scripts

Dogm, 2015-08-06:
also i dont see any avr-libc-1.7.2 version
In old folder max version:
* http://savannah.spinellicreations.com/avr-libc/old-releases/avr-libc-1.7.1.tar.bz2
new releases:
* http://savannah.spinellicreations.com/avr-libc/avr-libc-1.8.0.tar.bz2
* http://savannah.spinellicreations.com/avr-libc/avr-libc-1.8.1.tar.bz2

p.s. good traveling 8)

dcnewman, 2015-08-06:
On 06/08/2015 8:01 AM, Dogm wrote:
> did you mean this?
> http://www.netgull.com/gcc/releases/gcc-4.6.3/gcc-4.6.3.tar.gz
> http://www.netgull.com/gcc/releases/gcc-4.6.3/gcc-core-4.6.3.tar.gz
> http://www.netgull.com/gcc/releases/gcc-4.6.3/gcc-g++-4.6.3.tar.gz

Needs binutils (e.g., 2.22) and avrlibc (e.g., 1.7.2rc2252) as well.

Have you looked at

https://github.com/jetty840/Sailfish-MightyBoardFirmware/tree/master/dist/build-avr-gcc

Dan

Dogm, 2015-08-06:
http://ftp.gnu.org/gnu/binutils/binutils-2.22.tar.gz
http://ftp.gnu.org/gnu/binutils/binutils-2.22.tar.bz2

Dogm, 2015-08-06:
All ways avr-libc going to
* http://download.savannah.gnu.org/releases/avr-libc/
But there no 1.7.2 (only old-1.7.1 and new 1.8.0+)

Sample: 
* http://www.nongnu.org/avr-libc/user-manual/install_tools.html

Dogm, 2015-08-06:
It's dummy idea but maybe make VirtualBox/Wmware vdi?
Sample: http://www.osboxes.org/ubuntu/

It will contains stable compilation stand
I mean fixed scheme for all needed packages and versions

what do you think about?

dcnewman, 2015-08-06:
Good idea.

I use vagrant, virtualbox, and vmware fusion all the time, including using boxcutter, etc. to make compact, downloadable VMs.  So, I'm familiar with the concept.  I'll give it some thought.  Not something I can do until easily until next week when I return from my travels.   I may also just "make clean" and tar up the necessary tool chain packages -- they'll be easier for folks to download and build.

dcnewman, 2015-08-06:
Note that you can use avr-libc 1.8.  You just need to make the #defines's mentioned in the avr-gcc.mardown page,

__DELAY_BACKWARD_COMPATIBLE__ 
__PROG_TYPES_COMPAT__

Dogm, 2015-08-06:
also need update avrdude (config + win ver)
https://drive.google.com/folderview?id=0BxaDtEyC1MbnfnNERDBNU0R1YTk2UjhQN1hUb1NQOHhkal9rUjdlSlBIRFFMc1lkN3FSU0k&usp=sharing

Dogm, 2015-08-06:
I have programmer usbasp - which support in this build
+ this config allow use Atmega 16u2 chips
+ this allow use https://github.com/zkemble/AVRDUDESS

dcnewman, 2015-08-06:
1. I've updated build-avr-gcc-ubuntu12.04.sh to use gcc 4.6.3 and avrlibc 1.8.1.
2. I've introduced Compat.hh which for now detects avrlibc 1.8 and later and automatically defines some definitions to allow the old _delay(), prog_uchar, and prog_char to be used.  Compiles wiil toss warnings about prog_char and prog_uchar being deprecated.

Later, I'll build a fresh 12.04.5 VM in virtual box and do a full test.  Then, I'll see what can be done to make the box available.

Dogm, 2015-08-06:
8) thanks will test this...

dcnewman, 2015-08-06:
And I'm doing a

    vagrant package --base "Ubuntu 64bit 12.04.5"

at this moment.  The package.box will likely be ~2GB...  So, I may not get it uploaded anywhere until I return from my travels.

dcnewman, 2015-08-06:
I won't be able to upload the actual VM until after I'm done travelling.  Loading 1.2GB will just take too long.

Dogm, 2015-08-07:
hmmm why 1.2Gb?
maybe zip?

dcnewman, 2015-08-07:
That is compressed.  Standard trick of zeroing out the drive

sudo dd if=/dev/zero of=/EMPTY bs=1M
sudo rm -f /EMPTY

then shutting down the VM and letting Vagrant package and compress it.  They just aren't small.  (The .iso image itself for Ubuntu was 758M and it expanded to over 1.5G after being installed.)
that was with a variable sized Virtual Box .vdi for the virtual disk.

These things just aren't small.  It was the Ubuntu 12.04.5 64bit desktop install with only libreoffice* removed.  (Saves a little space.)  But all of gnome and base stuff was there.  Adds up.

Dogm, 2015-08-07:
why 64bit is need for compilers?
maybe just use x86 build (if it's smaller)?

dcnewman, 2015-08-07:
Habit.  I always only make 64bit VMs and never bother with 32bit.  I can try making a 32bit one and see if it's any smaller.  It will be smaller.  I just don't know how much smaller.