jetty840/Sailfish-MightyBoardFirmware #137: Additional localisations

Dogm, 2015-08-07:
Is need to add new language (Russian)?
I plane test and can publish localisation file if need...

dcnewman, 2015-08-07:
If you're willing to add Russian, that would be great.  However, I doubt the LCD screen supports Cyrillic.  At least not the ones we see here in the West.

Dogm, 2015-08-07:
Will see =)
it is possible only if no need rebuild/rewrite firmware for LCD (i don't want going there...)

Dogm, 2015-08-08:
Plane:
1) default build platform=wanhao_dup4 - done
2) write to chip (over icsp) - done
3) trying translations: - queued
3.1) cyrilic symbols - ?
3.2) translit symbols - ?

Dogm, 2015-08-08:
cyrilic test fail
https://monosnap.com/file/LOe0BbVgqj0ffww5pePTpekJVrMF3a

Dogm, 2015-08-08:
heh... seem so... need unsolder and rewrite chip rom with Russian symbols >)

Dogm, 2015-08-08:
translit - looks disgusting

Dogm, 2015-08-08:
found symbols (not tryed yet)
https://yadi.sk/d/XzcDSzzHiLiDr
later will try (need buy PLS-16 and PBS-16)

Dogm, 2015-08-08:
Hmm...
after inspect code for Mightyboard i found LiquidCrystalSerial libriary
maybe just make special firmware which can write special characters table for needed localization?

dcnewman, 2015-08-08:
You only get 8 special characters.  Looks like you can have 16, but not all the modules used actually seem to support 16.  If you put a symbol into position 8 you get the one in position 0.  I.e., you get index % 8.

Dogm, 2015-08-08:
what do you think about this?
https://github.com/mk90/LiquidCrystalRus

dcnewman, 2015-08-08:
It might work assuming there's available code space to fit it in. On a printer with an ATmega 2560, it should fit.  On one with a 1280, it may not.  Hard to say without trying.

Dogm, 2015-08-08:
i think this is bad idea for 1280 beacuse used Utf-8 (extra bytes for symbol code-page)
even if code size is enough - texts with custom codepage can be huge 8(

Dogm, 2015-08-08:
utf-8 mode fail =)
https://monosnap.com/file/fHLk96TDj9IKBiNudQvrK3KdkG9B2d
think no matter codepage...

Dogm, 2015-08-08:
hmm need to inspect
http://www.hobbytronics.co.uk/datasheets/HD44780.pdf

Dogm, 2015-08-11:
unfortunately impossible to make localization without replacement of LCD
8(

closed 2015-08-11 by Dogm