jetty840/Sailfish-MightyBoardFirmware #138: Idea: Some global variables

Dogm, 2015-08-11:
What if do some refactoring and place some board preferences as global variables?

I mean this:
----------------------
// microstepping is 1 / (1 << MICROSTEPPING)
...
//  4 for 1/16
//  5 for 1/32
...
#define MICROSTEPPING   4
------------------------

make define from scons script?
like: "stepdrv=a4988"
A4988 = 4
A4982 = 4
DRV8825=5
...
or just fixed set "stepdrv=4"

What do you think?

Dogm, 2015-08-11:
This affect
"firmware/src/MightyBoard/Motherboard/boards/<mb_folder>/Configuration.hh"
files

dcnewman, 2015-08-11:
Using 1/32 microstepping is pointless with these printers.  On the 16 MHz, 8bit AVRs the firmwares all quickly turn 1/32nd stepping back into 1/16th stepping by double stepping.  They do this because they are not fast enough to execute the necessary number of steps per second.  If you want to effectively use 1/32 microstepping you should be using an ARM processor.  Additionally, you cannot see any visible difference in print output between 1/16th and 1/32nd microstepping with FDM 3D printer.

Dogm, 2015-08-11:
sad 8(

closed 2015-08-11 by Dogm

dcnewman, 2015-08-11:
You have to keep in mind that one of the three key goals of sailfish is to get the highest performance possible out of the 8bit, 16 MHz AVR.  To accomplish that, hard assumptions are made about the underlying electronics and hardware including steps/mm for X, Y, A, and B around 100 steps/mm and Z around 400 steps/mm.   Also the approximate build dimensions.  If you start making a 1 mm Z axis (Jetguy) or high resolution stepping (e.g., 1600 steps/mm in Z) then issues can arise with the fixed point arithmetic.  Issues arise because the fixed point math can start overflowing.   So, our position has always been that if you want to do high resolution stepping or very large build volumes, then use Repetier or Marlin which both use (software) floating point arithmetic.  (And are about 5 - 6 times slower in the planner as a result.)

Other two key goals of Sailfish is safety and pressure/relaxation advance.

Dogm, 2015-08-12:
Understand...

Dogm, 2015-08-12:
Also ... what do you think about more powerful chips and progressive languages?
i mean what about try something like this
https://www.ghielectronics.com/catalog/product/543
https://www.ghielectronics.com/catalog/product/544
https://www.ghielectronics.com/catalog/product/530

Cpu: 120-240 mhz
Language: C#
Code: 1,3 ... 2,8 Mb
Ram: 10 ... 13 Mb
...
+ libraries for Usb/Lan/Display/....

Idea: do 3d printer software with this devices...

Dogm, 2015-08-12:
This assembly cost 150$ with graphic lcd display (tinker kit)
+ need board with 5-6 stepper drivers (can use any available interface)
and just power source ~10-25$

total cost - not so much... 
this allow to do more interesting devices with huge community of c# developers.
p.s. i think Arduino has smaller community even dot.net/c# developers..

toybuilder, 2015-08-12:
C# is not for real time embedded systems.

Dogm, 2015-08-12:
Does it matter if the reaction rate is clearly higher than in the real-time system?

toybuilder, 2015-08-12:
Yes. It does matter. I won't rehash the discussions, but if you look around
the various 3d printing groups, you will find plenty of past threads on
this issue.

The second answer on this link (
http://stackoverflow.com/questions/3762081/using-c-sharp-for-real-time-applications)
also captures the issues well.

Dogm, 2015-08-12:
I understand difference of both architectures...
may i ask of real problem with soft for 3d printer with non/real-time code?
if theoretically make such a system...

Dogm, 2015-08-12:
sensors? - it can have buffered with other chip currents

Dogm, 2015-08-12:
stepper drivers - it can update faster than delay of 1/64 microsteppings (for sample use standalone chip with stack of real time commands)

Dogm, 2015-08-12:
i just dont understand which of sub-logic must be only real time
explain to me please..

Dogm, 2015-08-12:
Also, if so antipathy witch non-real time languages and/or garbage collectors
just can use more powerful chips like
https://www.sparkfun.com/products/9713
* 80MHz
* Ram: 128KB
* Code: 512KB

dcnewman, 2015-08-12:
As I stated before, a key goal of Sailfish is to get the most performance possible out of 8bit, 16 MHz AVRs.  Sailfish isn't a firmware to port to a different architecture: it's not meant to be ported. And this isn't the place to be discussing new processors for 3D printing firmwares.  I do not know of a good place for such a discussion.  Perhaps 3dprintingtipstricksreviews?

Dogm, 2015-08-13:
I open discussion thread
https://groups.google.com/forum/#!topic/3dprintertipstricksreviews/aZtrkl9KPLE