jetty840/Sailfish-MightyBoardFirmware #139: Beeper Scream

Dogm, 2015-08-13:
Hi, 
please help to understand why my printer scream some time in printing?

Hardware: 
wanhao mightyboard rev.A

Software:
build ver: 7.8 (11/08/2015 get)
platform: wanhao_dup4 (with define+ 'ALTERNATE_UART')
language: en
all other options: defaults

Dogm, 2015-08-13:
scream when:
* print from SD
* print from PC (over alternate UART)

sound like end of print

dcnewman, 2015-08-13:
Could be your electronics: install the "official" firmware from the Sailfish release site and see if it still screams.  If so, you have an short in your electronics somewhere.  That's the typical cause of the piezo buzzer doing that.  If the problem goes away when you install an "official" sailfish, then it is a problem with your 11 Aug 2015 build of Sailfish that you have made.  Could be most anything including compiler/run time library bugs.

Dogm, 2015-08-13:
ok, i later test in official firmware...

Also it is possible if i use not official wanhao stepper drivers?
used A4988 (soldered ms2-ms3 pins)

dcnewman, 2015-08-13:
The buzzer going off when it shouldn't does happen with boards with electrical issues.  And the Geetech boards are not even "standard".  They contain electrical components which MBI dropped from the final design.  Components which may never have been tested (e.g., the safety cutoff circuit).

Dogm, 2015-08-13:
>>Hardware: wanhao mightyboard rev.A
not a geetech rev.e/g

Dogm, 2015-08-13:
i just short usb with wanhao board - and buy getech rev.g for tests
getech i dont use now (this problem with wanhao board)

Dogm, 2015-08-13:
this why i  use custom build (with optional uart)
i just solder PLS_4 pins on wanhao board and use it

i use simplify3d, it allow print with runtime show progress of buildings - cool feature...

dcnewman, 2015-08-13:
These boards are all rev E electronics.  Doesn't matter what internal revision number they put on the board.  They are all rev E.  Otherwise, Sailfish for rev E electronics would not work.

closed 2015-08-13 by dcnewman

Dogm, 2015-08-13:
hmm, talk about beeper is gone 8(

Dogm, 2015-08-19:
It just song after:
1) HBP warm done
2) Tool warm done
3) Raft build done

Why this going on? maybe this hooks in code?

dcnewman, 2015-08-19:
Keep in mind that this isn't a support forum.

Did you do as I suggested and install the "official" firmware build from the .hex file distribution site? Again the problem may be (1) your build, or (2) a short or other electrical problem with your motherboard.  With the official builds and for 4 years, no one else ever sees this except when they have an electrical short of some sort that is causing the buzzer to trigger under certain electrical conditions.

Dogm, 2015-08-19:
You right, default sailfish 7.7 work without this songs
but it dont have alternate uart 8( - i need this feature because 8u2 is dead...

Now i write this firmware (over isp)
<firmware major="7" minor="7" relpath="firmware/wanhao_dup4-Sailfish-v7.7.0-r1432.hex" name="Sailfish 7.7 (r1432)" description="Sailfish Firmware 7.7 (r1432) for Replicator 1 clones with 33.0 mm nozzle spacing. Requires ReplicatorG 40r28-Sailfish or later in order to connect via USB."/>

Dogm, 2015-08-19:
wanhao official board has unsoldered (but usable):
* AT1280 UART - i just solder PLS and it work perfectly
* AT1280 L2C
* AT1280 IO
* AT8U2 IO
* Q6 (Extra fan) - i just solder it and it work perfectly

But seem so i can't correctly build with current version in repo 8(

dcnewman, 2015-08-19:
... And the alternate uart support is only intended for ATmega 2560s.  There's not enough code space to correctly build it in for a 1280.  You're on your own.

Dogm, 2015-08-20:
ok sory for taken away time..

Dogm, 2015-09-07:
Hmm can you help find boards with ATmega 2560s compatible with Sailfish?

dcnewman, 2015-09-07:
Azteeg X3 (Panucatt, California)
Clone R1 (Carl Raffle, Switzerland)
FlashForge (China)

eckerj, 2015-11-09:
I know this may be a little late, but by "scream" you mean the piezzo buzzer makes loud noises after the print is completed?

If that's the case, it's because you have "Add celebration" enabled on the "Scripts" tab under "Post Processing" within Simplify3D.

I have a printer with a mightyboard and it is unable to correctly play any of the "tunes" available in Simplify3D.  

I do, however, have an Azteeg X3 with Sailfish installed and it plays all the tunes very nicely (though the volume is very very low).  My kids gave me the strangest look when they heard the "Looney Tunes" song play at the end of a print. :-)