jetty840/Sailfish-MightyBoardFirmware #141: what platform is I'm compiling?

burdickjp, 2015-10-08:
I have a FlashForge Creator Pro. I'm having difficulty finding in the documentation or through FlashForge what platform this would be considered. Am I ff_creatorx-2560 or ff_creator-2560?

dcnewman, 2015-10-08:
If you are compiling yourself and you wish to understand the differences between different platforms, then look in platforms.py, see what defines are established and then search for those defines in the sources.

If you are trying to figure out which publicly distributed binary to install, then know that there is only one FlashForge binary for an ATmega 2560 which is publicly distributed.  And that's absolutely the binary you need for a FlashForge shipped with an ATmega 2560.  Using anything else has an increased fire risk associated with it owing to the HBP FlashForge uses in those printers.

dcnewman, 2015-10-08:
P.S. If it shipped from FF with an ATmega 2560, the firmware build platform is ff_creatorx-2560 .

closed 2015-10-08 by dcnewman