jetty840/Sailfish-MightyBoardFirmware #143: Sailfish 7.8, Azteeg X3 Core-XY-min, Print Statistics menu selection causes Print Pause

eckerj, 2015-11-08:
I noticed that when selecting the "Print Statistics" menu option during a print actually causes the print to pause, rather than display the print statistics.

The "current_version.txt" file shows this at the top:
     7.08.0.1.0
      * Assorted bug fixes; Jetty/Dan ninth release

The firmware was built with:
     scons platform=azteeg-x3-xymin-corexy

The hardware is an Azteeg X3 2560 board with the original ViKi display/button board.

dcnewman, 2015-11-08:
First, realize that 7.8 isn't stable so problems are expected.  Do not use it in production.

Second, have you done a recent build?  This may be related to a known issue with the new menu item for turning the LED lighting on/off during an active print.  A counter associated with that menu was off and was fixed earlier in the week (or last week).  I haven't had a chance to test it yet myself.

eckerj, 2015-11-08:
Thank you for the feedback.   The one I have was downloaded Oct 17th or so.  I'll pull down the latest snap, build and report back.  Thanks again

dcnewman, 2015-11-08:
I just tried it on my Azteeg and things are working now.  The menu item for turning the LEDs on/off midprint still needs some more work though but otherwise things are working correctly now.

eckerj, 2015-11-09:
Thanks for humoring/educating me, Dan.  I grabbed the latest and greatest and it is indeed working now.  Thanks again...

closed 2015-11-09 by eckerj

dcnewman, 2015-11-09:
You're quite welcome.  And you nudged me to finish & test the printing-time menu item to turn the led lighting on/off (which some people who print in their bedroom/dormroom at night have been asking for).

eckerj, 2015-11-09:
Yeah, that really caught my eye!  Print time remaining! yay! :-)  Would that happen to be fairly accurate?  Even if it's not terribly accurate, it's still a welcome addition...

Thanks for all your hard work!

dcnewman, 2015-11-09:
The time estimates are only as accurate as the percentage markers added by the slicer (or GPX).
Enjoy!