jetty840/Sailfish-MightyBoardFirmware #147: Adding travis-ci support?

chrta, 2015-11-11:
Hello,

would it be interesting to add https://travis-ci.org/ support?

Only one file would have to be added and a travis account must be created (i think):
https://github.com/ArchimedesPi/Mendel90-Marlin/blob/master/.travis.yml

dcnewman, 2015-11-11:
I personally don't see the benefit, especially given that we use a very specific gcc toolchain that's no longer readily available.  We do that to eliminate problems caused by changes in the gcc toolchain.  E.g., to avoid the problem the Marlin folks had this past summer in which the newer gcc toolchains in Arduino 1.5 and later were causing issues with execution of their stepper interrupt code.  We saw similar problems circa 2012 when we upgraded gcc toolchains and ended up just nailing down to 4.6.2 and 4.6.3.

In my day job, we do run hudson and do continuous builds, automated unit tests, etc. so I am familiar with the practice.  I just don't see the benefit here.

dcnewman, 2015-11-11:
I looked a bit more: one would have to script pulling down the tarballs (binbutils, gcc-core, gcc-g++, libc), configuring, building, and installing them.  It can be done with

https://github.com/jetty840/Sailfish-MightyBoardFirmware/blob/master/dist/build-avr-gcc/build-avr-gcc-ubuntu.sh

but strikes me as potentially fragile.  This because I doubt the apt-get packages are available anymore for the 4.6.2 toolchain.  And that's the target toolchain for distribution of .hex files.  I suppose it could be automated using newer toolchains for continuous build testing, but I don't have any inclination myself to do that.

chrta, 2015-11-11:
Ok, i understand your point. I thought that travis could always build for every different platform and could check this way that no configuration breaks.

One question: Is the problem with other gcc and libc version documented somewhere? I only found https://github.com/MarlinFirmware/Marlin/issues/2370 I would like to try gcc 4.9.2 and avr-libc 1.8.0.

dcnewman, 2015-11-11:
Not really, no. We originally used 4.3 but then when we tried 4.4, we encountered problems (erratic motion for example). We tracked it down to what looked like a bug in one of the avrgcc floating point library routines. Contact with one of the avrgcc developers confirmed that and they told us to try the then new 4.6, warning us that the bug was in 4.5 as well. We did that and stuck with it until when MBI wanted to try 4.7 or thereabouts. More problems and several weeks of diagnosis didn't identify the issue. (MBI went off and spent a couple of weeks themselves.) Again, the problem manifested in some of the motion control. Anyway, we dropped worrying about it and have remained using 4.6.2 and .3 ever since. Moving to newer would require someone operating for a couple of months of use to provide assurance that there are not lurking problems. Even then, it would also warrant some timing/analysis of the stepper interrupt code to see what makes it the fastest. (We literally have some loops unrolled while keeping others because the structure led to the fastest timinings. Likely because it made the register checkerboards just right for optimization. Keeping the stepper interrupts -- there's two -- as fast as possible is important to overall performance.)

So upgrading compilers for distribution builds falls into the "if it isn't broken don't fix it category".  That is, there's no compelling reason to use newer compilers and libraries for the distribution builds so we've not worried the issue any further.  (Note that I do have a github repo with info on making your own VM with the 4.6.3 build chain.  That repo also has dropbox links to 32 and 64bit Ubuntu VMs with the toolchains installed along with their sources.)

chrta, 2015-11-11:
Thank you for this information.

closed 2015-11-11 by chrta