jetty840/Sailfish-MightyBoardFirmware #148: X/Y axis Jump when viewing LCD Menu.

KimBrown, 2015-11-26:
I have been printing a case to house the Ramps board on one of my printers here.
13Hrs in to the print, the monitor was telling me it was at 92% finished, so out of curiosity I wanted to look at the Total Statistics for the machine. In scrolling through the LCD menu I found what I wanted to see, and changed back to the monitor. When I glanced up at the nearly finished print, the X & Y axi had moved approximately 1 to 1.5mm to the right and Forward (i.e. X increase by 1mm & Y decrease by about 1mm) for no reason. The only thing that happend was I scrolled through the LCD Menu. I love SailFish, and am very happy with it, but that will be the last time I access the LCD when on a big print... So please be aware of the problem. I hope they catch the Bug Quickly.... (I'm using 7.6 at the moment, so maybe it's been fixed already in 7.7....I don't know)....

dcnewman, 2015-11-26:
There's no known bug in this area.  I suspect rather that you had an event which lost steps or you have the very common problem of a failing stepper motor cable.  FWIW, plenty of people, myself included, use those menus all the time during a print to turn fans on or off, set a new Z pause height, change the print speed, etc.

closed 2015-11-26 by dcnewman

dcnewman, 2015-11-26:
The one way I do know that people can cause this to happen is either if

1. They pause the print in which case the XY carriage is moved to the back right of the printer but 5mm away from each endstop.  HOWEVER, if at the start of the print they had inapt gcode which caused steps to be lost (e.g., running the XY carriage to the far front left of the printer and causing the motors to skip), then those lost steps can cause problems.  When the pause is done, the carriage may be moved back so far that it hits one or both of the endstops further losing steps and then causing a loss of registration.

2. With 7.6 and earlier, there was a bug whereby if you paused the print, did not have "heat while pause enabled", then loaded or unloaded filament, the X, Y, and Z stepper motors might be during the load/unload at reduced current thus making it easy to accidentally move the carriage while loading/unloading filament.  That was fixed in 7.7 but again requires that specific set of circumstances (actual pause, loading/unloading filament, and not having "heat while paused" enabled via Utilities > General Settings).

KimBrown, 2015-11-26:
Hi Dan, thanks again. That's interesting stuff. I'm 95% sure it was the filament jamming up now.I'm on a Large Harbour Tug Boat here, and with the noise might have missed the belts jumping as the carriage tried to move. I'll be home in about a week and will leave the long print till then to do.I've already updated to 7.7 before finding the problem.
As far as updating timing of the Update/Reset button, I found a good way to get it right.On the CTC and I'll guess the other clones as well, there is a green LED that flashes when there's comms going on on the USB. After pressing the UpLoad Button on the screen the Green LED flickers twice, with a very short ON duration (so appears dim), as soon as you see the second flicker, press the reset button. Works every time here...
Date: Thu, 26 Nov 2015 08:14:51 -0800

KimBrown, 2015-11-29:
Hi Dan, there seems to be a tiny Gremlin in 7.7 on the Preheat Settings, in that you can reduce the Heat Setting for the Extruder past Zero which then becomes 280'.... But going in the opposite direction you get stuck at 280'It's not what I'd call a Bug, just seems a bit odd.All the Best,Kim

Date: Thu, 26 Nov 2015 08:14:51 -0800

dcnewman, 2015-11-29:
Kim,

I checked in a code change that will limit the bottom-end settings to 0C for the pre-heat settings.  Thanks for reporting this.

KimBrown, 2015-11-30:
That's ok. Glad to help rather than be a hindrance. I used to do allot  of coding at one time, and know how easy it isto miss something like that.
I work on a ship (Large Harbour Tug bringing in Super Tankers to Oil and Gas Terminals here in Wales...), and going home in a couple of days, so the Printers packed up now in the back of the car, so doing this from memory.
I have a request for a mod in the software that shouldn't be hard to do if it's possible.
Off Sets:-In the Menu where you can set the Off Sets of the Nozzles would it be possible to set the Right Buttonto index throu from X, to Y, to Z (I haven't the machine here to check, but it was something I recall).[The Up / Down Buttons Inc / Dec the reading. Left moves out of the Off Set Option, Center ButtonSets the reading (If I recall), but the Right Button does nothing....
Also, only if there's space on the LCD screen for it, would it be possible to have it say which tool itrefers to. eg  'X for T(0)', or 'X for T(R)' .. I didn't play with this facility, but will need to soon as I have something a miss on my Left Extruder Off Set, unless I can alter it via the MakerBot or Rep software..When I saw it come up with a figure and it was only possible to move throughto the Y setting by 'Accepting' the X value on the screen, I thought I'd wait till I had a clear head and tucked up at home... It was an odd ball figure, so I wasn't sure if it was a reading it read from memory ora figure it picked out of the air something like 0.0018mm or something like that if I recall.If it read it from Memory, then it was probably a corruption that crept in when I tried upgrading fromone MakerBot version to another, before swapping over to SailFish.
Maybe it would be possible to have multiple presses of the Middle Button set the Off Set Reading that'sshowing to Zero.... Something like Pressing the Middle Button three times within 3 seconds or something similar.
I also need to find out what the PID number means for the Extruders. On my machine my two extrudersnumbers are different, and the only thing that is different is the rotation. But they might have got corrupted.(Guessing that the PID number tells the firmware the amount of Flow (quantity of filament to push through,maybe nozzle size and off set etc. and wondering if it's showing a Floating point number in place of a Binarynumber). I'll dig around on the internet and see what I can find out as I'm curious.
Thanks,Kim..

Date: Sun, 29 Nov 2015 12:22:12 -0800