jetty840/Sailfish-MightyBoardFirmware #156: Whas the last stabile source?

TraxXavier, 2016-05-08:
Hi,

I'm confused about what's the last stabile build of the firmware,
on the homepage it says version 7,7 a.k.a r1432 but this tag is not longer present in the repository.
So why is that, is there a new homepage? What version of the source should i build for productive use?

Cheers
Trax

dcnewman, 2016-05-08:
On 08/05/2016 1:21 AM, TraxXavier wrote:
> Hi,
>
> I'm confused about what's the last stabile build of the firmware,
> on the homepage it says version 7,7 a.k.a r1432 but this tag is not longer
> present in the repository.
> So why is that, is there a new homepage? What version of the source should i
> build for productive use?

That tag was never present in the repo.  Github is just how we make the sources
public.  We don't use it for actual development or building even.  Some time post
r1432, I started adding tags.

Dan

TraxXavier, 2016-05-08:
Ok
Is there a way to get the actual source for v7.7?

Also I failed to obtain the avr-libc 1.7.2 there is only 1.7.1 or 1.80 would you have by any chance a copy of the right libc, and could please upload it some ware?

Trax

dcnewman, 2016-05-08:
On 08/05/2016 10:14 AM, TraxXavier wrote:
> Ok
> Is there a way to get the actual source for v7.7?

The actual, full and complete sources are the github repo.  All that is missing is
internal notes (e.g., todo lists) and a couple of assorted tools -- scripts -- related
to signing packages and distributing (uploading) builds.

> Also I failed to obtain the avr-libc 1.7.2 there is only 1.7.1 or 1.80 would you
> have by any chance a copy of the right libc, and could please upload it some ware?

Please see

https://github.com/jetty840/Sailfish-MightyBoardFirmware/blob/master/docs/avr-gcc.markdown

Dan

TraxXavier, 2016-05-08:
Yes, but that does not answer my question:
what exact version of the sources in the git repo was used to build v7,7 a.k.a r1432
And where can I get avr-libc 1.7.2 the links provided in the *.sh files are all dead.

dcnewman, 2016-05-08:
On 08/05/2016 10:33 AM, TraxXavier wrote:
> Yes, but that does not answer my question:
> what exact version of the sources in the git repo was used to build v7,7 a.k.a r1432

Roll github back in time.  E.g., follow the commits, select one for about the correct
time, and then view the sources at that time and download the zip for that time.  Sailfish
7.7 was *officially* released 10 August 2014.  But r1432 was not the first, official v7.7
rXXXX.  r1432 was a bug fix to r1414,

https://github.com/jetty840/Sailfish-MightyBoardFirmware/commit/128cee9377c6f001f7da17e4742a1faf00cedc3e

You can use the above commit link to view the sources from that point in time and
with that download the zip of the sources from that point in time.

> And where can I get avr-libc 1.7.2 the links provided in the *.sh files are all
> dead.

As I previously wrote, please see

https://github.com/jetty840/Sailfish-MightyBoardFirmware/blob/master/docs/avr-gcc.markdown

Read what it says.  It points you at where you can get the avr-gcc packages.  Indeed, it points
you at VMs which have the entire avr-gcc build environment already installed.  And those VMs have
the zip files building and installing that avr-gcc environment.  You can move those zip files
to the environment of your choice if you do not want to use pre-built VMs.

Dan

dcnewman, 2016-05-08:
> And where can I get avr-libc 1.7.2 the links provided in the *.sh files are all
> dead.

Yes, the fact that they may be dead is actually addressed in that link I previously sent,

   The links in those scripts may no longer point at available downloads. The gcc maintainers
   eventually remove older tool chains from their repository thus invalidating the links.

That's why that document points people at VMs with the build environment installed and
.zip files for making/installing that build env on another x86/x64 system.

Dan

TraxXavier, 2016-05-08:
Well compiling the old source with the environment set up by the build-avr-gcc-ubuntu12.04.sh fails.
With the newest sources it works.

dcnewman, 2016-05-08:
On 08/05/2016 11:15 AM, TraxXavier wrote:
> Well compiling the old source with the environment set up by the
> build-avr-gcc-ubuntu12.04.sh fails.

Personally, I've never used the env. from those scripts.  They were
contributed by others.  With the environment in the VMs, things should
work with sources from summer 2015 forwards.  (I have never tried older
sources on those VMs.)

Dan

TraxXavier, 2016-05-08:
Whats new in r1605? if its semi final will there be a 7.8 final release soon?

dcnewman, 2016-05-08:
On 08/05/2016 12:12 PM, TraxXavier wrote:
> Whats new in r1605?

You'll have to look at the commits to see what has changed.  I've not had time to open
a formal field test for 7.8 and likely won't for a few more months.  When I do, we'll
go through the commits and see what has changed.  Mostly has been minor changes for
Core-XY, Azteeg electronics, and a new menu item to change platform temp mid print.
Owing to lack of program space in 1280's, there's not much room to add new features
for the majority of printers with Makerbot electronics.

Dan

TraxXavier, 2016-05-08:
> Owing to lack of program space in 1280's, there's not much room to add new features for the majority of printers with Makerbot electronics.

Yea, that is why I soldered a 2560 chip on mine.

New features I would like to have would be:
1. option to control a building space heater like you can the build plate, basically I need to read a NTC and control a relays that switches something like a hair dryer for example.
2. a switch that detects when the filament breaks, or more likely runs out so I can put in a new spool without the current print to break
3. may be: the encoder wheel one of the newest makerbots have is nice, it can detect if the filament feed fails, but that would be a larger HW mod.
...
..
.

Cheers
Trax

dcnewman, 2016-05-08:
> 2. a switch that detects when the filament breaks, or more likely runs out so I
> can put in a new spool without the current print to break
> 3. may be: the encoder wheel one of the newest makerbots have is nice, it can
> detect if the filament feed fails, but that would be a larger HW mod.

This has been supported by the firmware since April 2013. E.g.,

http://www.sailfishfirmware.com/v74-v44.html#pstop
https://groups.google.com/forum/#!topic/wanhao-printer-3d/iz3kDU7CQkY

There's even commercial hardware,

https://www.toybuilderlabs.com/products/tunell-3d-printer-filament-monitor

(P-Stop was actually added back in late 2012 but not officially released until 7.4
in April 2013.)

Dan

TraxXavier, 2016-05-08:
Ah great :+1: 
That is a really nice design...
I guess i will replicate it $65/piece is a bit very expensive...

closed 2016-09-11 by dcnewman