jetty840/Sailfish-MightyBoardFirmware #157: ff_creator-2560 platform should not have AUTO_LEVEL define

zapta, 2016-05-11:
Having the AUTO_LEVEL defined interferes with the display of the Z Pos in the stats page.

As far as I know, FFC do not come with auto level mechanism and thus there is no need to defined the  AUTO_LEVEL flag for the ff_creator-2560 platform. 

https://github.com/jetty840/Sailfish-MightyBoardFirmware/blob/master/firmware/src/platforms.py#L327

If some people want to experiment with auto leveling on FFC printer, they can use an alternative platform such as  ff_creator-2560-alevel such that the rest of the users are not penalized.

dcnewman, 2016-05-11:
Just because the FFC out-of-the-box doesn't have an auto-level probe doesn't mean that people don't use the firmware feature on FFCs.  Indeed, there have been several threads in the groups.google.com FF group of people using auto-level on their FFCs. Removing it from FFC builds would remove that feature for them.

closed 2016-05-11 by dcnewman

zapta, 2016-05-12:
Hi Dan, how about adding a platform definition without AUTO_LEVEL or alternatively, when the auto level mechanism is not available, not letting the "Auto-leveling inactive" message interfering with the Z pos? 

Vast majority of FFC users don't use auto leveling so optimizing it for their case seems very reasonable. If somebody goes through the trouble of adding auto level mechanism to FFC, updating the firmware to ff_creator-2560-alevel via Replicator-G isn't too much of a trouble.

markwal, 2016-05-12:
I disagree.  The cooling fan is also a feature that didn't come with FFCP for a long time, but many people added after the fact.  Having a proliferation of builds for every feature that someone might add seems unnecessary since carrying the feature on a 2560 costs you nothing (except for a menu item in settings that you don't click).  (IE I dont' think the rest of the users are penalized.)

dcnewman, 2016-05-12:
What I have considered is showing the Z height first, then leveling status when auto-leveling hasn't been activated.  Why not only show Z height when auto-leveling was activated for the print?  The reason to still show leveling status is in the event someone thought that their gcode had activated it but hadn't.  That way they will, after a few seconds, see positive confirmation that it's not activated.

markwal, 2016-05-12:
Or EEPROM setting?

zapta, 2016-05-12:
Yes, an EEPROM setting sounds good. I would make it off by default for the benefit of 99% of users that just use a stock FFC.  It's confusing to purchase a stock FFC from FF and getting that "Auto-leveling Inactive" message hiding the Z Pos.

Anyway, thanks for this great software.

zapta, 2016-05-12:
@markwal said: "... since carrying the feature on a 2560 costs you nothing (except for a menu item in settings that you don't click). (IE I dont' think the rest of the users are penalized.)"

In the case of the auto leveling, the "Auto-leveling inactive" message interferes with the ZPos display, so not just an unused menu entry.

markwal, 2016-05-12:
@zapta, Unfortunately, I think you'd have to default the other way just for the least change on upgrade.

dcnewman, 2016-05-12:
@zapta said: "It's confusing to purchase a stock FFC from FF and getting that "Auto-leveling Inactive" message hiding the Z Pos."

I don't see how it might be confusing: either they've read the documentation and thus understand how it works, or they've not read the documentation and thus don't even expect to see the Z Pos the first time.

Regarding the EEPROM setting, I have no plans of adding one for this: the cost of adding EEPROM settings is too high and thus need a significant justification (e.g., a not insignificant amount of code work in RepG is associated with each new EEPROM setting).  I just don't see this as one.

As an aside, if you mostly use only a single extruder you can try setting the extruder count to 1 and then power cycling the printer.  You should then get, during a build, the running ticker on the main screen of the build time so far, estimated build time remaining (based upon the % milestones put into the gcode/x3g by the slicer), the filament used so far, and the current Z height.