jetty840/Sailfish-MightyBoardFirmware #158: Inconsistent ZPos label

zapta, 2016-05-12:
When compiling without AUTO_OFF, the stats page starts with ZPosition label and then switch to ZPos. 

Looking at the code, I think it's:

https://github.com/jetty840/Sailfish-MightyBoardFirmware/blob/master/firmware/src/MightyBoard/shared/Menu.cc#L3280

vs. 

https://github.com/jetty840/Sailfish-MightyBoardFirmware/blob/master/firmware/src/MightyBoard/shared/Menu.cc#L299

Suggestion: change the line at first link to use the same message as the second link, and if Z_POSITION_MSG is not used in other places, just drop it.

dcnewman, 2016-05-12:
Addressed in https://github.com/jetty840/Sailfish-MightyBoardFirmware/commit/85ae77aa0f0225501ff1024335f5cc8fe5b085f5

dcnewman, 2016-05-12:
Thanks for catching and reporting this

closed 2016-05-12 by dcnewman

zapta, 2016-05-12:
Thanks Dan.