jetty840/Sailfish-MightyBoardFirmware #161: v7.7 sometimes computes accelerations incorrectly?

dbavatar, 2016-06-23:
Is this a known bug? I've noticed this in two different brand replicator clones, and using multiple slicers, although it was unclear what was happening at first. Basically max speed change XY seems to work *almost* all of the time, but for certain arcs, it seems to have minimal effect, hammering the print head instead, sometimes skipping steps. This happens in both axes, and continues to happen even if changing the onboard speed multiplier from 1.00 to 0.35. If I turn down max speed change XY to 1 or 2 mm/s, then I observe it does have an effect on the entire print (slowing down arcs that printed fine before), and pretty much eliminates the hammering. But 5 for example, is too high! If I try and manually jog with acceleration disabled, it isn't a problem for my printer to go much much faster (i.e. over 30 mm/s)

I do have test objects that do this now kind of reliably. A 6mm circular "hole" in an object seems to do it, or a block with decreasing radius holes.

dcnewman, 2016-06-23:
Known issues with all firmwares when printing small circles.  Often discussed in various 3DP forums.  You address this with your slicer.  E.g., slic3r's slow down for small detail.  Or the old RepG "altshell" plugin which has an option to slow down for all interior loops as well as all interior and exterior shells.

At issue is centripedal acceleration, v^2 / radius.  The firmwares have no clue that they are printing an arc.  All they see are line segments one after another.  And all they limit speeds for are linear accelerations.  Since they do not realize they are doing a fast circle or curve, they do not know to limit the speed so as to limit the centripedal acceleration.   And it's particularly bad when you have excessive fine detail (e.g., 6 mm diameter circle with 300 facets): in that case, the angle change between successive line segments is very small and so the linear accelerations are very small.  Thus the firmware allows the segments to run at nearly top speed.  But since the centripedal accelertion scales with the square of the linear velocity, that really makes matters bad.

So, again, you have to address this in your slicer.

closed 2016-06-23 by dcnewman

dbavatar, 2016-06-23:
But interestingly sometimes an arc accelerates properly, and sometimes the same arc in a different layer hammers, in the same print, obviously with the same slicer settings. In fact, the time it takes to get around the circle seems relatively constant, hammer or none. So to me it seems either like a bug, or some missing implementation. I would expect to it hammer every single time over a 6mm circle if it was a straightforward limitation as you say. Regardless, keeping some sort of time ordered data like a small histogram or even simpler a moving average vector with exponential decay over multiple segments shouldn't be a big deal, computationally. That would fix the v^2/radius problem, at least.

dcnewman, 2016-06-23:
> Regardless, keeping some sort of time ordered data like a small histogram or even simpler a moving average vector...

On a 1990's 16 MHz, 8 bit processor with no floating point and only 4K of RAM of which there's about only 300 bytes free.....  And a processor which doesn't have much in the way of spare compute cycles left either.  These 1990s AVRs are actually pretty asthmatic when used for 3DP doing acceleration planning, heater management, stepper control, I/O control to a LCD, possible I/O with a USB chip (and which must be done at interrupt level owing to the lack of flow control to the USB chip), etc.

No, I would not expect it to always be identical each and every time since the processing isn't deterministic: external events are constantly impacting the stepper interrupt (e.g., reacting to temp changes and having to do PID calcs when needed; reacting to serial I/O from the USB chip at interrupt level; etc.)  Indeed, Sailfish and Marlin actually pull back the speed when they find that external events are compromising the ability of the stepper interrupt to keep up: this prevents situations where the printing might otherwise actually pause for a millisecond or two.  Such pauses can (and do) cause little zits which are visible to the human eye when they appear on an outer shell.  So, that's one specifc case where printing may go differently from on otherwise identical loop to the next one.  Another example is in how slicers may order the printing leading up to the loop from layer to layer: that too can impact the speeds run on the loop and make them different from layer to layer.

As to being a firmware issue/bug, again all the 3DP AVR firmwares have this issue.  Again, it comes up somewhat routinely in all the forums -- questions about this jerky, rough, slamming behavior on small circles and arcs.  So if it's a bug, it's shared by G3Firmware, G4Firmware, Sailfish, Marlin, Grbl, Sprinter, Repetier, and Teacup.  (Possibly others, but those are the AVR firmwares I routinely follow.)

dcnewman, 2016-06-23:
Also, with the typical Replicator 1, Replicator 2/2X, and clones thereof, having a stepper motor cable develop an intermittent break is quite common and can lead to odd, intermittent behavior involving jerking, stuttering motion.  Not likely to be the case here, but a possibility nonetheless.  Also, if the digipots are not receiving clean signals from the processor, then the motor currents may not be being set correctly which can lead to lost steps at higher speeds.  (On the Asian clone printers, you should never bother dropping the currents during warmup -- there's no reason to and it admits the possibility of them not raising again. The Asian clone printers all have 14.6A PSUs which obviate the need for dropping currents.  Not that it ever needed to be done anyway; MakerBot could have heated things up AND then homed rather than homing first heating second.)  However, this isn't a support forum: such issues are better addressed in whatever forums might exist for your particular brand of printer.

dbavatar, 2016-06-25:
I believe I found the problem, it has to do with how slowdowns on buffer underruns are handled. It's none of the problems above. I'll do a pull request assuming I don't find other problems with it, but on my test gcode it's working very well both for circles and as I discovered, even straight lines that have a lot of segments. The only thing I haven't figured out is that moves_queued seems to have a wrap issue (less than zero), although it's mysterious how that happens. I don't suppose I can get a serial console? 

https://github.com/dbavatar/Sailfish-MightyBoardFirmware/tree/hammerfix

dcnewman, 2016-06-25:
Interesting.  Very briefly, when the speed is reduced for low buffer, it should be more gracefully increased when the buffer is healthy again?

Now, I would hope that accel planning would take care of that as it imposes kinematics, but it isn't (or isn't sufficiently)?  I haven't had time to take a careful look.

The wrap issue with moves_queued: it's a circular buffer and wraps around.

dbavatar, 2016-06-25:
Ahh, ok now that makes sense. Well, assuming the ring buffer is filled above the threshold again on the next segment, it will go back to full speed immediately with my fix, but if it's only populated within the slowdown area, then it will still have a proportionately moderated speed, but that's a good thing.

dcnewman, 2016-06-25:
And, FWIW, here's a dark secret of the Grbl-derived accel planners (Older Grbl, Sprinter, Marlin, Repetier, and Sailfish among others): the kinematic constraints are imposed in the "initial" planner, plan_buffer_line().  But then the iterative planners, planner_reverse_pass & planner_forward_pass, all ignore the constraints other than the initial rate at the start of the segment.   They try to maximize the speed and can and do violate the max speed changes.  Worse yet, in many of the Grbl derivatives, calculate_trapezoid_for_block() doesn't ensure that entry_factor and exit_factor do not exceed 1.  When they exceed 1, which they can do on occasion, the initial or final rate can be scaled to be larger than nominal_rate which is the previously computed max speed for that segment.

I've not looked recently at the recent work in Grbl so this may no longer be true for Grbl.  Hence my writing "older Grbl".