jetty840/Sailfish-MightyBoardFirmware #163: Toolhead offset of 0 doesn't actually make the toolhead offset 0

Zakm0n, 2016-06-30:
I'm running a mixing extruder, so I need the toolhead offset to be exactly 0, but whenever I set it to 0, it still prints at 33. Setting the offset to -33 moves the offset -66 from the original spot. Is there something in the firmware that just defaults an offset of 0 to the factory default?

dcnewman, 2016-06-30:
Any X offset in the range -4 to +4mm is interpreted as a deviation from ideal (e.g., 33.0mm).  See the fifth paragraph of [Section 9.2 of the Sailfish Reference Manual](http://www.sailfishfirmware.com/doc/details-toolhead-offsets.html#x50-1270009.2).   This is how the MakerBot-style firmwares accommodate a change in behavior MakerBot introduced circa 2011: they used to store the toolhead offsets as the deviation from the ideal gap but then in v6.2 (IIRC) they switched to using the actual offset rather than the deviation from the ideal offset.  The firmwares attempt to accommodate EEPROM with the old settings (deviations) as well as new settings.

What can you do?  You would have to do your own firmware build commenting out or otherwise defeating this test

[4mm test](https://github.com/jetty840/Sailfish-MightyBoardFirmware/blob/master/firmware/src/MightyBoard/Motherboard/Steppers.cc#L342)

And then build.  Easiest way to build is to use one of the pre-built Ubuntu virtual machines mentioned in the first bullet item [here](https://github.com/jetty840/Sailfish-MightyBoardFirmware/blob/master/docs/avr-gcc.markdown).

closed 2016-06-30 by dcnewman

Zakm0n, 2016-06-30:
I ended up checking the box "Apply toolhead offset to g-code" in Simplify3d and setting T1 offset to -33. Gcode preview is messed up, but it prints perfectly.

dcnewman, 2016-06-30:
I nearly asked you which slicer you were using since I know that for at least RepG 39 and S3D you could try that trick.  Nice to have a workaround, albeit with an annoying artifact.