jetty840/Sailfish-MightyBoardFirmware #169: JKN Advance K and K2 for Bowden extruder conversion

caall99, 2016-08-15:
I recently converted my Qidi Tech Dual printer to a dual Bowden extruder. I have been trying to dial in JKN Advance K and K2 to reduce the amount of filament extruded during acceleration and deceleration. It seems I get appreciable print quality above 1.5 for K, and 0.055 for K2. For the sake of it, I tried a value of 10 for K and saw NO difference. It leads me to believe that after a certain value it has no effect. What is the maximum value? My Bowden is still extruding too much in corners and at the end of every infill pass... even at a K value of 10. What else can I do to improve my situation?

dcnewman, 2016-08-16:
On 15/08/2016 4:00 PM, caall99 wrote:
> I recently converted my Qidi Tech Dual printer to a dual Bowden extruder.

With Bowdens, you need to have massive retraction settings (e.g., 6 - 8 mm).  At that point,
you're outside the scope of applicability for JKN advance which was intended for direct
drive extruders.  The K likely does work with Bowdens, but the heuristic relief (K2) isn't
meant for Bowdens.   If you have ARM electronics, you should consider RepRapFirmware as
it has a pressure advance similar (nearly identical) to the K portion. But it also has
a portion intended for Bowdens.

Dan

caall99, 2016-08-16:
Would you say that sailfish firmware is not suitable for bowden extruders? If changing the k value no longer has an impact, how can i pull in my corners and generally reduce the amount of filament extruded during acceleration and deceleration? What arm board would be suitable to upgrade my Replicator 1 Dual clone to?

dcnewman, 2016-08-16:
On 15/08/2016 6:15 PM, caall99 wrote:
> Would you say that sailfish firmware is not suitable for bowden extruders?

Some people do use Sailfish with Bowdens.  I do not know how successful they are or are not.
Sailfish is highly optimized for a particular style of 3D printer.  Part of that design is
direct-drive extruders.  It was not optimized for Bowdens.

? If changing the k value no longer has an impact, how can i pull in my corners and generally
> reduce the amount of filament extruded during acceleration and deceleration?

Larger retraction in your slicer, not slowing down as much at corners, etc.  Same as
any other printer really.  The blobs are caused by (1) excess pressure, and (2) slowing
down too much thereby allowing more of the ever-present ooze to build up as the corner
is turned.

> What arm board would be suitable to upgrade my Replicator 1 Dual clone to?

The Duet 0.8.5 and Duet WiFi are nice boards.  They both run RepRapFirmware.
The RADDS board can run Repetier and RepRapFirmware.  (However, it doesn't have
ethernet or a built in web interface like the Duet's do.)

Dan

caall99, 2016-08-18:
Hey Dan,

Thanks again for all your input. So I have the Bowden setup working
reasonably well, but am still fighting some blobbing at the Z scar, and
differing rates of extrusion, most notably during acceleration and
deceleration.

My firmware/Simplify3D settings are as follows (this is using a dual Bowden
setup - stock hot ends/nozzles, just relocated steppers to back of printer
- using a 2mm ID PTFE tubes for 1.75mm filament):

X/Y Accel: 800 mm/s^2
Z Accel: 150 mm/s^2
A/B Accel: 6000 mm/s^2

Retraction Distance 7.00 mm
Extra Restart Distance: -0.25 mm
Retraction Vertical Lift: 0.00 mm
Retraction Speed: 80.0 mm/s
Coast at End: OFF
Wipe Nozzle: OFF

Print speed: 80 mm/s (but reduce to 40% of print speed for outlines)

I retract at every layer change.

Do you suggest that I increase X/Y acceleration, to reduce blobbing in
corners, and/or increase the effectiveness of K and K2? I have thrown
arbitrarily high values for K, (e.g.. K=14) and am not seeing much of an
impact. What are maximum values for K and K2?

caall99, 2016-08-29:
JKN values ended up not being the problem. I resolved this issue with other tweaks... Remaining issues with Sailfish will be commented on elsewhere.

closed 2016-08-29 by caall99