jetty840/Sailfish-MightyBoardFirmware #171: Small prints on v7.8 don't finish correctly.

jasoroony, 2016-10-17:
Some jobs I use to calibrate my CTC machine just draw a square (openscad 0.2mm high cube with the difference of a 2mm smaller cube inside, leaving a 1mm square outline - or basically 8 lines to draw).

The machine gets to about 5~6 lines done and stops, doesn't lower the Z axis all the way down like it does on much larger jobs.  The X/Y max parking doesn't always go all the way either, both go a little distance and just stop!

This works properly on v7.7.

I've noticed the Z (bed) descend at max speed too once on v7.8 at job end (only once) and when my thermocouple wire broke (giving an intermittent reading for a while) the first extrusion line of each new Z layer also went at full travel speed.  I'm not sure if it was the job or the broken wire or a bug, but I've never seen these things on v7.7.

Can I also suggest InterfaceBoard.cc : getUpdateRate test for 3 HostState signals : HOST_STATE_BUILDING & ..._FROM_SD & ..._ONBOARD and divide the rate by 5 or so if the machine isn't busy in those states - to give the user a nicer experience in the menus while setting up the machine?

Another idea is an option to limit the Command.cc : HOST_COMMAND_DISPLAY_MESSAGE handler so the timeout_seconds isn't set above a configurable value (3 seconds would be my choice) since when the job finishes and I want to print another I'm left stuck on the message screen for 5 seconds when I really want to restart the heaters quickly!

Finally, is the azteeg_x3 config actually confirmed/known to work still in the latest v7.8?  I've been playing around with it for another partially built machine and while the menus/interface etc. seems OK, along with repG config control, the stepper movement is not working right for me.  Moving any axis in repG (X, Y or Z) only retracts X if I'm lucky...  The jog menu makes a click sound and stops (but I think it's just running really really slow because I sometimes hear a click of the motors at a very slow consistent interval).  Oh, also, the SD card menu seems to run really slow too - very very sluggish but still works :)  These things might be due to my machine though, so I'm just asking!

Thanks for listening!

jasoroony, 2016-10-17:
Oh, nearly forgot, ThermocoupleReader.cc : reset : 2nd for loop has <= TEMPSENSORS which goes out of the array bounds.  Needs to be only "<" like the first for loop.

jasoroony, 2016-10-17:
No wait!  There's one other thing...  For anyone else out there wanting to build the Sailfish source code on a Windows machine with the Arduino toolchain and Windows Python...

SConscript.mightyboard should have 3 lines changed:

  	'-DVCS_VERSION_STR=\\"'+str(vcs_version).zfill(5)+'\\"',
	'-DVERSION_STR=\\"'+version_str+'\\"',
	'-DDATE_STR=\\"'+datetime.now().strftime('%y/%m/%d')+'\\"',

Otherwise, you will need to use linux, not that there's anything wrong with linux - other than the 64bit kernel doesn't thunk 32bit client code to 64bit kernel operations for all required DTV tuner card io-controls...  A pet hate of mine, since code has been offered up but ignored for years by those "in control".

dnewman-polar3d, 2016-10-17:
Whose 7.8 builds are you using?  The sort of issues you describe are the sort of problems which can turn up if you change compiler versions.

As to the Azteeg builds, the current 7.8 Azteeg build for Core-XY has been working fine for me for the past month with no difficulties.

Jasoroony, 2016-10-18:
I first noticed it on the http://jettyfirmware.yolasite.com/resources/beta/firmware.xml for v7.8 r1675 that I downloaded on 2016/05/14.  That's yours, right?  Compiling v7.7 source I downloaded recently (branch 2014/08/07) with the tool-chain I have works as well as the official v7.7 (non-beta, from the same site) that I used for almost a year previously.  Reason being, I wanted to make some small changes (that I've described above) and they work for me with this older version well enough now.  The tool-chain seems fine (though I do blame my tools too when things go wrong ;).

I've totally reset the EEPROM on the azteeg too now, my last thing to try, but still no luck - it's not a Core-XY though.  It's just like the CTC except that it's an XYmin machine.  I'm still wondering why my changes to the menu speed don't make the SD card's long filenames scroll across the LCD far faster like it does on the CTC.  Maybe the SD card is slower - but I assumed that was controlled by the Arduino clock.

There's something wrong going on, Marlin and Repetier work on the mechanics/electronics of the new machine without any issues at all.

FYI - I also have a 24x8 LCD that I tried with Sailfish - but only the main menu seems to handle the extra lines without doing the wrong thing - I think the general settings screwed up the most.  It'd be nice if it worked.  FYI 2 - There's also some warnings (unreferenced parameters) still present in v7.8 when I disable DIGIPOT_SUPPORT when I tried different drivers that doesn't need it.

Anyway, got to get off this git thing now... ghosting out.

dnewman-polar3d, 2016-10-19:
Yes, if it's from [http://jettyfirmware.yolasite.com/resources/beta/firmware.xml](http://jettyfirmware.yolasite.com/resources/beta/firmware.xml), then I did it.

As to Azteeg, I had to break down and use `-mrelax` to link it.  One of the vtables was crossing some boundary.  That owing to code changes.  We've absolutely avoided doing that until now and I can readily believe it to be a cause of issues on the Azteeg.  But if you're seeing the same issues on other machines, then I'm far less certain.

Jake Bordens at one point put some work into getting > 4 line LCD displays working but I don't believe he ever finished it.  Unfortunately, I have zero free time for the next 3+ months.

Jasoroony, 2016-10-22:
Issues: (Azteeg & Rep1)

1.
X moves when X,Y or Z is told to move when PLATFORM_EXTRUDERS=1
Everything works much better when there are two extruders defined.

2.
C++ Constructor madness, change a few things:
extern const AvrPort should use PortA={0x20}; etc. for A-L

And use this instead of the class (since Sailfish makes no real use of it)

typedef struct _AvrPort {
  port_base_t port_base;
} AvrPort;

coolingFan shouldn't reset() in the constructor, it's done elsewhere.

heater shouldn't reset() in the constructor, it's done elsewhere.

(These have to do with accessing other classes in a constructor but
C++ doesn't always "construct" in the order you are expecting.)

3.
pstop_incr() needs conditional compiling on #ifdef PSTOP_SUPPORT

4.
sync on certain commands: last added one: HOST_SET_BUILD_PERCENT
causes small jobs to end before they are complete. (20 Oct 2015 commit)

5. (Needs more thought)
XY_MIN_HOMING & startHoming(command==HOST_CMD_FIND_AXES_MAXIMUM
I've changed this to false on condition of XY_MIN
This needs more thought, but I want my XYZ MIN machine to work
with what makerware produces and it assumes XY MAX, Z MIN.

6.
liquidcrystalserial 'down' bitmap is a different format (7x8) to the others.

7.
SettingsMenu::drawItem shouldn't use % 4, but rather % LCD_SCREEN_HEIGHT

(The enter EEPROM screen doesn't 'clear' more than 4 lines either.)

8.
SDMenu::SDMenu should use a new class bool "underConstruction" = true,
then call reset(), then set to false...
SDMenu::resetState should use itemCount = 0; when "underConstruction" else use
itemCount = countFiles().

(This prevents a C++ constructor from initialising the entire SD library.)

9.
stepperaxis defines pstop_port when PSTOP_SUPPORT isn't defined.

10.
steppers DEBUG_SLOW_MOTION max_speed_change[B_AXIS] set when EXTRUDERS=1

11. (Opinion only)
thermocouplereader should have an error_code for each sensor, like temp[].

12. (FYI)
With the C++ constructor bugs fixed, Sailfish now compiles/works properly with the Arduino IDE.

dnewman-polar3d, 2016-10-23:
Thanks.  I've made nearly all the changes (and some earlier ones).  I'll check them in in a bit after I have a change to test them a little.

1. Building with PLATFORM_EXTRUDERS 1 was an experiment done at one point for the Monoprice Architect, a single extruder printer running Sailfish.  However, Monoprice only did a limited run of 500 or so of that printer and they put the standard two extruder Sailfish on it.  So, work didn't go any further on supporting PLATFORM_EXTRUDERS 1.  (That said, it works in the Gen 3 code base for the Cupcake, but that's a much older Sailfish code base.)  Net, net, not surprised to hear there may still be bugs when PLATFORM_EXTRUDERS is 1 for the mightyboard code base.
2. Pretty ancient code dating all the way back to the Gen 3 and possibly Gen 2 code base.  (Maybe even Gen 1?  Older source code repo was lost by RepRap when things cut over from svn to git late 2010 or thereabouts.)  I made your changes.  I had looked at that code in the past and thought about pruning it as yet another space savings.  Thanks for working it through.  Also took care of those extra reset()'s.
3. Taken care of.  Missed noticing that when ZYYX sent that change.  Again, thanks.
4. Unsure of what you're driving at.
6. Not sure if it makes a difference?
7. Got it, thanks.
8. Good point, thanks.
9. Got it, thanks.
10. And thanks again.
11. Have to think about that.
12. Yikes.  Reason lots of folks do NOT use the Arduino environment is the extreme lack of care shown by the maintainers as regards release noting behavior changes.  Too many people have been bitten by their changing behaviors and not bothering to warn anyone.  Becomes a matter of either going over each release carefully, or just sticking with a fixed release.  (Recent example: repeated changes in WDT handling in their SAM3X8E support which bit a number of different projects finding the WDT working, then not working, and then a significant change in how to enable.)

dnewman-polar3d, 2016-10-23:
Just put in a bunch of changes.  Thanks much.  (Commit comments all cite you.)

Jasoroony, 2016-10-23:
No worries, thanks for making my printer better, and hello from Australia.

4.
That's the first git commit that makes my small jobs stop before they finish completely, building the previous snapshot works...  However...

While removing the if test in the latest release also makes my small jobs finish properly, it somehow introduces a tiny stutter ! (regardless of the gcode having the build percent update in there or not) and that occurs regularly on all other jobs causing blobs.  I'm not sure if that previous snapshot did it because it was only something I noticed after I deleted the older code.

Hand crafting some gcode to test, based on the Makerware output, sending the head around the HBP in a rectangle at various speeds, I notice it gets further along the faster it's going.  Timing the point when the print-another menu appears (which happens on these small jobs before the job finishes, before the tune is played/finished-message is displayed) to when the tool stops unexpectedly is *always* around 30 seconds.  If I send the tool around fast enough, it all completes properly.  Since my calibration tests are all first layer extrusions, they need to be slow.

Could just be a 30 second timeout (that I couldn't seem to find anywhere) which doesn't affect bigger jobs because the machine is going much faster after the first layer.

5.
Without this, my machine (which has no MAX endstops) sends the tool smashing into the rail-holders (or it would if I let it).  With it, jobs created for my CTC will run on the new machine too.  At least, with the machine in its current state, it looks good to go.

12.
Sure, I understand many libraries are in flux in the Arduino environment, but all the 3D printer firmwares available except Sailfish work with it and it makes it very easy for others without coding experience to use it.  Since it's all open source, most firmwares including Sailfish have their own copy of libraries known to work with their code too.  What little Sailfish uses outside of its source base, IMO, is unlikely to be changed to to the point where it breaks Sailfish.  You would also be likely to see all the 3D printer firmwares experience simultaneous issues if it did happen - so no big deal.

I'm also sure, with a little bit of effort, you could make Sailfish still work with your scons build process *and* also work with the Arduino IDE.  Best of both worlds.  Just need to change *.cc to *.cpp, *.c to *.cpp, *.hh to *.h and add a nice "config.h" to each .cpp file containing all the build type #defines that scons is doing at the moment which you conditionally exclude if scons is detected by a new define.

Interesting to note, the latest Arduino IDE / GCC (4.9.2) release creates smaller code and I have not seen any issues with the calculation engine (yet) having trouble keeping up with my CTC machine speeds that are set quite high.  I.E. the 4.6.2 / 4.6.3 Sailfish issue.  I'm probably not in a position to judge though, I've never seen the issue to know.. and I've upgraded to TMC2100 drivers on X/Y, which are so quiet, I might not notice.

One thing I noticed though when playing with -Ofast (instead of -Os) was isqrt1() causing assembler issues due to it's assembler labels being repeated throughout the optimised code `( Sqrt16, _sq16_1..4 )` which I worked-around by adding: `static int16_t __attribute__ ((optimize("O0"))) isqrt1(...).`  But that change should be conditionally compiled in only if further optimisations are applied over -Os.  Pretty sure -O3 also did it.  Alas, the CTC doesn't have enough flash for -Ofast, but I'm also not convinced of it needing any more performance with the current options.

13.
I noticed something else I fixed before, there's a couple of #if HAS_RGB_LED in Sailfish that should really be #ifdef's.

Jasoroony, 2016-12-25:
The ghost of Christmas past says pipe_clearing = true; if ((command!=HOST_CMD_SET_BUILD_PERCENT) || ((command_buffer.getLength > 1) && (command_buffer[1] == 100)))

dnewman-polar3d, 2016-12-28:
Are you referring to this?

dnewman-polar3d, 2016-12-28:
Are you referring to this?

    % svn diff Command.cc
    Index: Command.cc
    ===================================================================
    --- Command.cc	(revision 1681)
    +++ Command.cc	(working copy)
    @@ -1493,7 +1493,7 @@
   			    (command != HOST_CMD_FIND_AXES_MAXIMUM) &&
 			    (command != HOST_CMD_TOOL_COMMAND) &&
 			    (command != HOST_CMD_PAUSE_FOR_BUTTON) &&
    -		    (command != HOST_CMD_SET_BUILD_PERCENT)) {
    +		    ((command != HOST_CMD_SET_BUILD_PERCENT) || ((command_buffer.getLength() > 1 &&) (command_buffer[1] == 100)))) {
        	                         if ( ! st_empty() )     return;
        	                 }

Jasoroony, 2016-12-29:
Looks good to me.