jetty840/Sailfish-MightyBoardFirmware #176: CTC Rep1 Dual - Extrude or retract on second toolhead causes an X movement?

C44Supra, 2017-01-31:
Hi I've been running Sailfish v7.7 r1432 on my CTC Bizer / Replicator 1 Dual for quite a while now. I've never had any problems until I tried adding the second hotend into the mix. Every time I send the second toolhead (T1) an extract or retract command, it causes the machine to move it's X position. It happens with the machine controls in ReplicatorG and with the machine controls in Octoprint (haven't tried anything else yet).

The machine otherwise runs absolutely fine, toolhead T0 (right side) works as it should. T1 however will always, no matter if I extract or retract, move the X to the right (positive).

dnewman-polar3d, 2017-01-31:
This is normal.  It's how it's supposed to work.  To change tools, it needs to move the extruder to the left or right approx. 34 mm.  And the rule you're supposed to follow in actual gcode is change tool, retract the filament.  That way, the lateral motion is combined with a printing move -- it's combined with the retraction move.

closed 2017-01-31 by dcnewman

C44Supra, 2017-01-31:
Well.. eh. That makes sense. But wouldn't that mean that the toolhead should move left once you switch back to T0? (Still rather new to using the second toolhead).