jetty840/Sailfish-MightyBoardFirmware #178: Heating LED not really Red

SergeantFreeamam, 2017-02-10:
Hi!

Is there an option in the firmware, where i can set my own clours to specific temps?
When my printer heats up with activated heating LEDs its even at 220°C quite blue (only a bit purple).
I have searched in the firmware for over 2 hours now.....
I cant even find the script for the heating LEDs.
Thanks!

dcnewman, 2017-02-10:
On 10/02/2017 6:08 AM, SergeantFreeamam wrote:
> Hi!
>
> Is there an option in the firmware, where i can set my own clours to specific temps?

No

> When my printer heats up with activated heating LEDs its even at 220°C quite blue (only a bit purple).
> I have searched in the firmware for over 2 hours now.....
> I cant even find the script for the heating LEDs.

https://github.com/jetty840/Sailfish-MightyBoardFirmware/blob/master/firmware/src/MightyBoard/Motherboard/Motherboard.cc#L545

Dan

SergeantFreeamam, 2017-02-10:
Thank you! 
thats it!