jetty840/Sailfish-MightyBoardFirmware #180: Mighty_one Board cannot heat the extruder heater after upload 7.8 (r1682)

ThanaphanIn, 2017-03-13:
I do build Sailfish-mightyboard firmware from version 7.8 (r1682). 
scons platform=mighty_one
After upload this firmware to Mightyboard Geeetech Rev G board.,the overheat light on the board is show up. I cannot do Preheat or any printing .
The current board is working with Firmware 7.7 (r1432).
Please recommend me. What I do wrong?
-Thanaphan In.

dcnewman, 2017-03-13:
On 13/03/2017 1:14 AM, ThanaphanIn wrote:
> I do build Sailfish-mightyboard firmware from version 7.8 (r1682).
> scons platform=mighty_one
> After upload this firmware to Mightyboard Geeetech Rev G board.,the overheat light on the board is show up.

geetech board?  It's not even a correct Mightyboard.  Geetech used the wrong schematics.  However,
the overheat LED is always lit.  It's part of an incomplete circuit not populated on a real
Mightyboard.

> I cannot do Preheat or any printing .
> The current board is working with Firmware 7.7 (r1432).
> Please recommend me. What I do wrong?

You're on your own.  Sorry.  If you choose to build the firmware yourself, it's up to you to diagnose any issues.

thuSkywalker, 2017-04-18:
> geetech board?  It's not even a correct Mightyboard.
Just to make sure. Did you mean that the real Rev. G looks like this while Geeetech Rev. G is still similar to Rev. E?
https://i.imgur.com/aSxheya.jpg