jetty840/Sailfish-MightyBoardFirmware #181: Slic3r with FFCP and sailfish extruder select failing

mron, 2017-04-29:
I can't get Slic3r and gpx to play well. When selecting Sailfish as the gcode flavor, Slic3r inserts`M108 T0` or `M108 T1` for tool changes. These don't seem to work. The printer always selects T1.

Should an `M108 Tx` change tools?

dcnewman, 2017-04-29:
On 29/04/2017 8:22 AM, mron wrote:
> I can't get Slic3r and gpx to play well. When selecting Sailfish as the gcode flavor, Slic3r inserts|M108 T0| or |M108 T1| for tool changes. These don't seem to work. The printer always selects T1.
>
> Should an |M108 Tx| change tools?

Look at the gcode generated by slicers which work then emulate what they do.  Namely, look
at the gcode generated by ReplicatorG 40r34-Sailfish.   And note the tool is not changed until
a motion command next occurs.  The motion of the tool change is added to the next motion command.
That, of course, doesn't work in general but it's how MakerBot did it.  So, the rule is that
after a tool change the next motion command should only move A or B but not X, Y, or Z.  That
way when the X and Y displacements of the tool change are added in, the motion of the extruder
nozzle if extruding will not be wrong in XYZ space.  In practice, MakerBot would do the tool
change and then a slight retraction of the filament in the extruder nozzle just switched away
from, IIRC.

Dan

natewin, 2017-04-29:
You may also have better luck just outputting RepRap flavor and letting GPX do it's thing.  I think i recently read a post in Google Groups suggesting that.   It is what I have been doing from Slic3r Prusa edition and my tool changes just worked.

markwal, 2017-04-29:
Somebody convinced Allesandro to emit "M108 Tn" for toolchange for "Sailfish" flavor, but it's really only ReplicatorG flavor and not all ReplicatorG's at that.  Henry Thomas didn't want M108 to be toolchange (he refused to add it at some point).  I actually added it to recent versions of GPX, but I still recommend avoiding it because it means other more important things to other firmwares like "interrupt wait" on Marlin.

You can workaround this in Slic3r by putting "T[next_extruder]" in the tool change gcode box and then you'll get both the M108 Tn followed by a Tn by itself, but actually, I recommend doing what @natewin said:

Set your output flavor in a Slic3r to "RepRap" and use the default mode of GPX to convert to x3g (no flag or "-r" if you want to be explicit).  In fact, I recommend using "RepRap" flavor wherever possible, so the "MakerBot" flavor should only be for slicers that only know that flavor (ReplicatorG, MakerWare and MakerBot Desktop) all of which know how to make their own x3g's.

mron, 2017-04-29:
I did try RepRap. Tools change, but the bed tries to heat to the Extruder temp. Current solution is use Sailfish and preprocess gcode, replace M108's with T's then send to gpx

markwal, 2017-04-29:
Make sure you set Reprap in *both* the slicer *and* GPX.

mron, 2017-04-30:
Thanks, I have both extrudes and the correct temperatures for extrudes and bed working. I switched to your version of gpx and I'm using reprap gcode flavor. There is an X misalignment between the extruders I looking into now.

markwal, 2017-04-30:
By how much?  34mm?