jetty840/Sailfish-MightyBoardFirmware #185: Missing platform

FelixTremblay, 2017-10-27:
So, from what I can see, there doesn't seem to be an option to build the firmware for a wanhao duplicator 4 with an atmega2560. I planned on changing the chip on my printer to a 2560 in the next week or so to enable auto-leveling but since this configuration doesn't seem to exist, which steps would I need to take to build a working firmware for my machine? I'm guessing that only a couple of build options need to be changed/added but some help would be appreciated. Also, I mistakenly bought NO inductive sensors and I do believe that the default z-endstop behaviour is to be NC. Which build option do I need to change to also fix that?

FelixTremblay, 2017-11-19:
I have soldered the atmega 2560 on the board, flashed the regular bootloader (sk500boot v2) and uploaded a couple of firmwares to the chip. My problem right now is that, even though I can access menu options by pressing the navigation buttons, the screen blank. Actually, it isn't blank... Everything on the screen is lit. Would anyone have an idea on what the problem is?

dnewman-polar3d, 2017-11-19:
1. You cannot use the "regular" bootloader.  It must be built with some additional compile-time options.  Use [this one](https://github.com/dcnewman/MightyBoardFirmware-2560-bootloader/tree/master/goodHexFiles) which is built with the correct compile-time options.
2. Be sure to use the correct firmware.  You will need to use a build for a Replicator 1-style printer and NOT for a Replicator 2 or 2X.   (I run across more and more people who believe they have a Replicator 2 clone when they really have a Replicator 1 clone.  Doesn't help any that CTC even advertises some of their printers as Replicator 2's.)  If you use a Rep 2 or 2X build, then the LCD display won't work -- it's driven by different output pins on the 2560.

FelixTremblay, 2017-11-19:
Thanks for the quick answer. About the bootloader, by "regular", I meant the sk500boot v2 atmega2560 used by the flashing script in the dist directory. I do think that it is the same that you linked. for the firmware, I made a custom entry in the platforms.py file based on the regular wanhao dup4 1280 firmware and changed some things to make it 2560 compatible. Here's the code I added : 
```
   'wanhao_dup4-2560' :
        { 'mcu' : 'atmega2560',
          'programmer' : 'stk500v2',
          'board_directory' : 'mighty_one',
          'defines' : [ 'HEATERS_ON_STEROIDS', 'COOLING_FAN_PWM',
                        'PLATFORM_SPLASH1_MSG=\\\"Sailfish Wanhao Dup4\\\"',
                        'PLATFORM_THE_REPLICATOR_STR=\\\"Wanhao Duplicatr\\\"',
                        'PLATFORM_TOOLHEAD_OFFSET_X=3201',
                        'PLATFORM_X_OFFSET_STEPS=13763L',
                        'PLATFORM_Y_OFFSET_STEPS=6919L',
                        'HAS_RGB_LED', 'EEPROM_MENU_ENABLE', 'ALTERNATE_UART' ]
        },
```
I also rechecked the chip this morning to make sure that there are no solder bridges and that every pin is fixed to its pad.

dnewman-polar3d, 2017-11-19:
You can always compile it such that it will drive the debugging LED bank (which you need to connect up).  You'd add 'DEBUG_ENABLE' to that list of defines in above.

For a Mightyboard,

    #define DEBUG_PIN       Pin(PortB,7)
    // Additional Debug Pins
    #define DEBUG_PIN1      Pin(PortC,1)
    #define DEBUG_PIN2      Pin(PortC,0)
    #define DEBUG_PIN3      Pin(PortG,1)
    #define DEBUG_PIN4  Pin(PortB,7)
    #define DEBUG_PIN5  Pin(PortG,0)
    #define DEBUG_PIN6  Pin(PortK,6)
    #define DEBUG_PIN7  Pin(PortK,5)
    #define DEBUG_PIN8  Pin(PortK,4)

The low bit of the debug output will be output on C1 and the high bit on K4.  And the LED combinations are

    High three bits indicated which module
    
    LED #
    876
    000 -- Main::
    001 -- Motherboard::
    002 -- Steppers::
    
    Low five bits are startup stage within the module
    
    Main::
    
    Deci LED #
    mal- 54321
     1 - 00001 -- Main::main() started
     2 - 00010 -- Motherboard::init() finished
     3 - 00011 -- Main::reset() started; interrupts locked out
     4 - 00100 -- WDT enabled in Main::reset() [only appears on hard reset]
     5 - 00101 -- sdcard::reset() finished
     6 - 00110 -- Piezo::reset() finished
     7 - 00111 -- utility::reset() finished
     8 - 01000 -- command::reset() finished
     9 - 01001 -- steppers::init() finished
    10 - 01010 -- steppers::abort() finished
    11 - 01011 -- steppers::reset() finished
    12 - 01100 -- motherboard::reset() finished
    13 - 01101 -- Main::reset() finished
    
    Motherboard::
    
     1 - 00001 -- Motherboard::init() has been entered
     2 - 00010 -- SoftI2cManager::getI2cManager().init() finished
     3 - 00011 -- interface::isConnected() finished
     4 - 00100 -- Motherboard::initClocks() finished
     5 - 00101 -- Motherboard::init() about to return
     6 - 00110 -- Motherboard::reset() entered
     7 - 00111 -- UART::getHostUART().enable() & .in.reset() finished
     8 - 01000 -- InterfaceBoard::init() finished
     9 - 01001 -- InterfaceBoard::pushScreen( splashScreen ) finished
    10 - 01010 -- Finished 3 second display of splash screen
    11 - 01011 -- InterfaceBoard::init() finished
    12 - 01100 -- TWI_init() finished
    13 - 01101 -- RGB_LED::init() finished [only for RGB builds; write to I2C bu    s]
    14 - 01110 -- Piezo::playTune( startup tune ) finished
    15 - 01111 -- Do nothing cutoff.init() finished
    16 - 10000 -- Tool 0 reset() finished
    17 - 10001 -- Tool 1 reset() finished
    18 - 10010 -- HBP thermisitor init() finished
    19 - 10011 -- HBP reset() finished
    20 - 10100 -- RGB LED colors asserted [only for RGB builds; writes to I2C bu    s]
    
    Steppers::
    
     2 - 00010 -- stepperAxisInit() finished
     3 - 00011 -- initPots() finished

That will at least tell you how far the motherboard is getting in initializing itself.   I suspect it will get all the way and that the issue is with the LCD display?  But it could be something else.

FelixTremblay, 2017-11-19:
Oh god am I feeling like a complete dumbass right now... Guess who Is too fu***ng retarded to inspect a chip correctly? That price goes to me! LCD-data pin wasn't quite soldered on. To my defense, that side of the chip is the only side I tried to do with solder paste/hot air and I am not as too experienced with the technique. I am so sorry for wasting your time. Thanks for the help.

FelixTremblay, 2017-11-28:
So, I saw in the build options that there is a PLATFORM_ENDSTOP_INVERT option. Is this the right option to modify if my sensor is NO and right now, the platform only goes up if the sensor is triggered? I tried modifying it by making it : 'PLATFORM_ENDSTOP_INVERT=0b10011011', but it doesn't seem to be doing anything. I even tried to invert all endstops by doing 'PLATFORM_ENDSTOP_INVERT=0b10011000' but none of the endstops were inverted. Am I looking at the right build option?

FelixTremblay, 2017-11-29:
UPDATE : For some reason, the eeprom wasn't updating itself when I was flashing the firmware. I dumped the eeprom on the sd card and manually edited the fifth byte in the bin file. I don't know if this is a software issue or if it is just because I don't know how to properly flash the printer so I'll keep that issue open until further notice.