jetty840/Sailfish-MightyBoardFirmware #186: Use B extruder as primary

nicoco007, 2017-11-25:
I have an old MightyBoard Rev H with blown circuitry next to the A extruder output which prevent the hotend from receiving power. However, the B extruder ouput still works. Is there any way I could build Sailfish to use the B extruder by default instead of A?

dnewman-polar3d, 2017-11-25:
Yes, you would just swap all two heater pin assignments between the two extruders in the Configuration.hh file (motherboard/boards/mighty_two/Configuration.hh).  Then rebuild the firmware -- not an easy task -- then install it.  Hook the extruder's heater wiring to the B extruder's and you're then good to go.

See [https://github.com/dcnewman/Ubuntu-Sailfish-VM](https://github.com/dcnewman/Ubuntu-Sailfish-VM) for some virtual machines you may be able to use to set up a toolchain with which to compile the firmware.

nicoco007, 2017-11-26:
Alright, I'll give it a shot. Thanks!

nicoco007, 2017-11-26:
I switched all the A/B combinations in the configuration file, build it in one of your VMs, and it works like a charm! Thank you very much for your help!

closed 2017-11-26 by nicoco007

dnewman-polar3d, 2017-11-26:
Congrats!  And thanks for the followup.  It is appreciated.