jetty840/Sailfish-MightyBoardFirmware #187: Need to be able to turn the endstops off for auto-level with inductive sensor.

FelixTremblay, 2017-11-30:
I just finished setting my printer with an atmega2560 and installed a fresh copy of sailfish on it. I made a somewhat decent start gcode script but right now, the printer isn't working as expected. During the first layer, the extruder slowly gets farther from the build plate. I'm pretty sure that this is due to my inductive sensor always being triggered during the first layer. The printer is able to adjust the print height by going down a bit, but when it comes to going back up, it thinks that the printer is already at its max height so it keeps it from going down. It seems like the M121 command should disable the endstops but it doesn't seem to be implemented into sailfish.

dnewman-polar3d, 2017-12-01:
MakerBot printers do not implement commands to disable endstops.  Such commands were introduced in rep rap firmwares to work around build problems.  E.g., making endstops which are too susceptible to EM noise and then bundling the endstop wiring with the motor wiring.  Rather than then fixing the real problem (wiring + endstop electonics), Rep Rap firmwares implemented commands to simply disable the endstops.   MakerBot's philosophy has always been to fix the underlying problem.

FelixTremblay, 2017-12-01:
Oh, I see... So, what would you recommend to do in my case? Would there be any way for me to just "ignore" the endstop after the homing process? Or is there an other I/O I could use to plug in the inductive sensor? If I could check the status of the sensor while the bed is going up, wouldn't I be able to stop the z movement and to store that as a point? 
As a last resort, I might be able to make a holder that positions the sensor slightly in the front left of the extruder so that I could mesure the points while being off the plate. That way, I could position the sensor slightly higher and it wouldn't trigger while printing.
It would kinda be a shame to be that close from having a auto-leveling bed but to be blocked by a software limitation like that :/
I really tried understanding the code to modify it but my first-year college student skills still aren't enough for modifications bigger than changing some pin assignments...

dnewman-polar3d, 2017-12-01:
> Oh, I see... So, what would you recommend to do in my case?

Change the sensor to only trigger when you're using it as a zprobe.

> Would there be any way for me to just "ignore" the endstop after the homing process?

No.  You'd have to modify the source code.

> Or is there an other I/O I could use to plug in the inductive sensor? 

No.

You have to keep in mind that Sailfish is meant for MakerBot style printers.  It is heavily optimized for the specifics of MakerBot style printers to the extent of not supporting broad deviations from the design (e.g., doubling the steps/mm for each axis).  If you wish to use an inductive probe for the Z axis, then you need to switch to a firmware intended to support inductive Z axis probes.  Sailfish is not.

FelixTremblay, 2017-12-01:
Found a sketchy solution so I'd like to get some feedback on if this is a good solution or if I might just burn my house down. I don't use my right extruder so what about I just remove the fet controlling the extruder heater and plug my inductive sensor in there. At its max, it takes 9mA which is all right for an atmega2560 pin. From there, I can just plug a random thermocouple to make sure the "heater" is allowed to be powered on and give power to the sensor with a regular M104 command, which could be disabled when I actually start the printing process. This is sketchy as I do need to make sure that the sensor is powered on when I home the machine but I feel like it could work

dnewman-polar3d, 2017-12-01:
Won't work.  The printer, when a heater is enabled, watches the rising temp and will trigger an error if the temp isn't rising.  Which it won't be since there's no heater heating the temp sensor.

FelixTremblay, 2017-12-01:
How about that : I don't use the right extruder. I broke the driver and decided it would just be easier to remove the right motor. What I could do with that is to use the DIR pin of the broken stepper driver (the driver has been removed from the printer so I would only need to plug stuff into the header) and I could turn on/off the inductive probe by "extruding" in a certain direction. the pin seems to keep the last state it was in so it would be easy to control the sensor.

FelixTremblay, 2017-12-02:
[Progress update]

The DIR pin doesn't actually keep its last state. I managed to pull something off which worked with a multimeter but when I connected the actual probe, the nozzle just crashed into the build plate. This seemed like a great start but it just feels way to unreliable since my sensor would be in an off state most of the time so I'll go an other way.
 
Right now, the easiest path seems to be to just take the cooling fan output, but the fact that the FET is after the fan terminal creates a 24V DC offset that makes it hard to work with. I could take the pin output directly at the FET, but then I'd probably lose the ability to have a software controlled cooling fan, which I need.

My next idea is to combine this project with a project that is further in the todo list : setting up octoprint with the GPX plugin. I feel like I might be able to control the power line of the sensor directly from the raspberry pi running octoprint. I'll try this and post an update later just in case there is an other person who is crazy enough to do this upgrade in the future.

FelixTremblay, 2017-12-12:
[Almost a final update]

The octoprint approach seemed great at first, but the gpx plugin just isn't powerful enough and created some unwanted limitations (I'm not blaming the plugin author here, he has to work with the limitations of the machine itself). 

I decided to go with the cooling fan output. To overcome the DC offset and to make sure that I would be able to to still use a cooling fan, I wired up an optocoupler in parallel with the fan to switch the probe on/off.

This also creates some limitations. As an example, with the current height of the probe, I can't start my cooling fan until the fifth layer, which is a bit higher than usual for me. I also have to have my fan on when probing the plate. The air from the fan might be able to cool the plate a bit, affecting slightly the precision of the probed point.

I will leave the printer in its current state until I find some time to work on it again but I have an idea about using an arduino to send only a pulse to the fan at before probing. The arduino could differentiate the small pulse from an active on state and allow greater control over the state of the probe.

uoitalia, 2018-06-17:
Look at my release AMD code