jetty840/Sailfish-MightyBoardFirmware #191: Generate EEPROM map for MakerBot Desktop?

rpavlik, 2017-12-28:
The files at the ReplicatorG-Sailfish "thing" are fairly old, and so I'm not confident that they match the current "tip-of-tree" firmware I've built and installed. And, despite the README there saying it's integrated, MakerBot Desktop 3.10.0.1364 complains it doesn't have EEPROM maps for my device (a Replicator 2X, with upgraded sailfish firmware) if I try to edit parameters.

It looks like there is markup to automatically generate the files required, however, the only script I see in this repo takes in a CSV file, not a header file.

Any advice on how I can do this? (Or, alternately: it doesn't look like the eeprommaps file has changed much, perhaps I can just minimally modify the 7.7 file to identify itself as 7.8?)

rpavlik, 2017-12-28:
Ahh, it appears related to the EepromAnalyzer.py tool included in the makerbot software. I'll still accept hints, but I'll also accept if you want to close this as not-a-bug (though documentation would be handy)

dcnewman, 2017-12-28:
That EepromAnalyzer.py was some half-finished stuff MakerBot wrote back in the MakerWare days.  It never worked well.  I'd fire it up, get it to run, get a map, clean it up, and send it to MakerBot.  I stopped doing that when they renamed MakerWare to Desktop and made it clear that they no longer wanted to support anything other than their 5th Gen printers.  (Since they still were selling the Rep2X they reluctantly had some support for it in Desktop.)

FWIW, here's a URL for a copy of [`makerbot_driver/EEPROM/EepromAnalyzer.py`](https://www.dropbox.com/s/joofudedky6h63e/EepromAnalyzer.py?dl=0) I have from the s3g-surprise repo.

But no real hints.  MakerBot seemed content with the tool in a so-so state.  And I didn't want to expend any of my time cleaning it up given that they might completely change their map files without notice.

closed 2017-12-28 by dcnewman