jetty840/Sailfish-MightyBoardFirmware #192: Request - Backlash compensation?

alfi666, 2018-01-11:
Hi all, 
first of all thank you for putting this firmware together - works a lot nicer than original.
There is one improvement though that would be very helpful - and that is backlash compensation.

As most of you know, backlash is the distance between motor changing direction and the tool actually moving to that new direction.

From software perspective, this would require basically just adding a variable that IF the direction of movement change THEN move the motor by X amount of steps to new direction on top of original steps. 
If calibrated correctly, this can completely remove any backlash.

There are some slicers which are capable of doing this, but most of them are not :(
Is there anyone who could implement this right into the firmware?
BR, Peter

dnewman-polar3d, 2018-01-11:
Sailfish is designed for a very narrow set of 3d printer designs.  And meant to perform as well as possible on those specific designs.  That specifically includes not attempting to fix mechanical design errors in software.   The Sailfish view of this would be that you need to correct the hardware issue at the hardware level, not the software level.

alfi666, 2018-01-11:
Hi, thanks. I cannot agree - because sailfish is fixing the HW issue of mass being thrown around by acceleration/deceleration, which is basically overcoming of a HW issue as well :) Implementing the "correction" on movement change is pretty simple compared to that.
The G code even has a commands for that (M98 and M99) but they are often not implemented in the FW.
I reckon, best would be to have a flawless HW, but backlash correction is pretty standard in CNC world so I'm a bit surprised it is almost non-existent in 3D printing world.

dnewman-polar3d, 2018-01-11:
Mass being an insurmountable issue, barring some form of inertialess drive, rates addressing in software.  Backlash, OTOH, has proven to be addressable at the levels needed for 3DP via careful hardware design.  However, it's open source.

alfi666, 2018-01-11:
Sure, if I'd be capable of wrapping my head around the source code I would do it myself, however it is way more complicated than what I could cope with.
I'm looking into options of modifying the G code to compensate for the backlash, however having it in firmware would be much nicer :)