jetty840/Sailfish-MightyBoardFirmware #193: Fan Speed

Rockevzel, 2018-02-12:
Hi there, 
I hope, I am right here - I've never been on Github before. Please have mercy mith me :-D 

I can't find out: Does this Firmware already include the Fan-Speed Control for the 'Extra'-Fan?
Or is it only: On/Off ?

If not: Is there a way to programming the speedcontrol? Many ppl - including me, are waiting for the solution :-D 

Thank you :)

dnewman-polar3d, 2018-02-12:
X3G does not permit a fan speed.  It's just ON or OFF.   Some people have modified their gcode to X3G translator AND Sailfish to allow for a variable fan speed.  But there's no plans to incorporate that into Sailfish.  Well, at least not by me as I no longer have to to test and make new releases.

The current Sailfish 7.8 code base has code to allow you to set via the LCD screen a fan speed (0 - 100%).  That speed will then be used when the fan is turned ON via a print.  So, it's a "static" fan speed as opposed to a fan speed which can change during the course of a print.

Rockevzel, 2018-02-12:
Ahh I see.. do you know who did this translator and the modified sailfish ?

I bring you my printer for testing :-D the whole community hopes for an
update :-D

dnewman-polar3d, 2018-02-12:
Look at outstanding pull requests on this repository.