jetty840/Sailfish-MightyBoardFirmware #194: Calibrate nozzle offset problem?

rpavlik, 2018-02-13:
I've had this happen a few times, with recent builds and after resetting to factory defaults each time.  I can go in and follow the instructions here: http://www.sailfishfirmware.com/doc/tuning-dual-extruder-calibration.html 

But, when I'm done (after choosing, e.g. line 4 for x - so -0.3mm expected offset), looking at those settings in ReplicatorG shows a number in the 3000s, and it's not a replicatorg viewing issue, because if I try to print then with the second extruder the print carriage will cheerfully and alarmingly smash itself into the left-hand side of the gantry (presumably trying to get to 3000...)

I haven't tried a proper dualstrusion print after manually editing that value in EEPROM using ReplicatorG to what I'd expect (-0.3mm), but I have tried ditto printing which does work successfully (as long as I remember to turn it off when done... :man_facepalming: ).

dnewman-polar3d, 2018-02-13:
Sounds like you have a mismatch between RepG and the firmware you are running.  You MUST use both RepG 40rXX - Sailfish (for some value of X) downloaded from the Sailfish Thing at thingiverse.com and actual Sailfish firmware.  If you use the wrong RepG you'll see incorrect values.  (And if you use MakerWare with the Sailfish EEPROM maps, you'll see the values in units of steps which is how they are stored in EEPROM.)

Worst comes to worst, you can set them via the LCD UI: Settings > Toolhead Offsets.  Then you can set the values to whatever you want.

rpavlik, 2018-02-13:
I am using the most recent one available in binary form. However, even if I never touch repg except to turn the calibration print from gcode into x3g, I still get the "bang the head against the wall" issue. I don't recall actually even seeing them in makerware despite figuring out how to generate the updated eeprom maps :-/.

dnewman-polar3d, 2018-02-13:
Well, keep in mind that there's a garbage in/garbage out aspect to this calibration: the system adjusts based upon your CURRENT toolhead offsets.  That is, that LCD UI in which you enter the line indexes just does a delta to your existing toolhead offsets.  So, if they were wildly off beforehand then they will likely remain wildly off.  You may want to force them to what you know to be the ideal values and then do the calibration.  For a genuine Rep 2X, it is 35.00 mm for X and 0.00 mm for Y.  So, you can forcibly set those via RepG (or the LCD UI).  Then do the calibration.  If after doing the calibration and using the LCD UI to enter index numbers you then again get wonky values (e.g., 3000), then something is indeed broken.

rpavlik, 2018-02-16:
I followed the instructions that I linked to in the docs, which involved setting them to 0 beforehand using repg-sailfish. I believe that's what's called the "new model" where the fixed/nominal extruder distance is built in to the firmware so 0, 0 would be "ideal printer"...