jetty840/Sailfish-MightyBoardFirmware #195: Sensitivity to file names on SD card

rpavlik, 2018-02-13:
I'm running a build based on the latest master on my Rep2X, where I've gone in the platforms.py file and modified the mighty_twox entry to be as follows (that is, disabling "squeeze", and adding ` 'RGB_LED_MENU', 'S3G_CAPTURE_2_SD=1', 'ACCEL_STATS=1', 'SD_RAW_SDHC=1' ` to defines because I want those features and they'll fit in flash/memory...)

```python
    'mighty_twox' :
        { 'mcu' : 'atmega1280',
          'programmer' : 'stk500v1',
          'board_directory' : 'mighty_two',
#          'squeeze' : [ 'Menu.cc', 'Interface.cc', 'InterfaceBoard.cc',
#                        'LiquidCrystalSerial.cc', 'DigiPots.cc',
#                        'Eeprom.cc', 'EepromMap.cc', 'Piezo.cc',
#                        'UtilityScripts.cc', 'TemperatureTable.cc',
#                        'Thermistor.cc', 'Thermocouple.cc', 'Heater.cc',
#                        'CoolingFan.cc', 'PID.cc',
#  '[ os.path.basename(f) for f in glob.glob(\'../../src/MightyBoard/Motherboard/boards/mighty_two/*.cc\') ]' ],
          'defines' : [ 'BUILD_STATS', 'HAS_RGB_LED', 'COOLING_FAN_PWM',
                        'PLATFORM_SPLASH1_MSG=\\\"  Sailfish Rep 2X   \\\"',
                        'PLATFORM_TOOLHEAD_OFFSET_X=3100',
                        'PLATFORM_THE_REPLICATOR_STR=\\\"Replicator 2X\\\"',
                        'PLATFORM_MACHINE_ID=0xB017',
                        'PLATFORM_X_OFFSET_STEPS=13463L',
                        'PLATFORM_Y_OFFSET_STEPS=6643L',
                        'PLATFORM_AXIS_LENGTHS={246L, 152L, 155L, 100000L, 100000L}',
                        'PLATFORM_AXIS_STEPS_PER_MM={88573186, 88573186, 400000000, 96275202, 96275202}',
                        'EEPROM_MENU_ENABLE', 'RGB_LED_MENU', 'S3G_CAPTURE_2_SD=1', 'ACCEL_STATS=1', 'SD_RAW_SDHC=1' ]
        },
``` 

This is better since adding SD_RAW_SDHC - seems less picky - but some files on an sd card will not show up in the menu or in the list on my attached OctoPi/Octoprint setup, and it seems to be related to their filenames. I've tried many cards, all non-HC cards, have clean formatted them in a number of ways [camera, rufus, official sd formatter], and have tested them and removed the faulty [original MakerBot] one as well as the 256MB card that everything except the replicator can read from my collection.

General rules I've found so far: The `-` character in filenames (especially if there are multiple - say 3 or more) is almost certain to make a file not appear in the list (though `extrude-test-standalone.x3g` works fine, somehow). Too many `_` characters will also do it, and length also plays a role.

For instance, here's the file I was having issues with at the moment that prompted me to finally file this bug. The Cura export from `Soldering_Fingers_Holder.stl` is called by default `MR2X_Soldering_Fingers_Holder.x3g` - and that won't show up. Length also plays a role, seemingly, since after removing some underscores to make it `MR2X_SolderingFingersHolder.x3g` (renaming in place on the card or copied to a different card with that name) it still won't see it. Finally, when renamed to `MR2X_SolderingFingers.x3g` it shows up in OctoPrint as being on the printer's SD card.

I'm not sure how best I can supply information on this issue. The card I have in there right now has the following files showing up in OctoPrint:

```
mr2x_aaholder3.x3g
mr2x_bp5l-housingbase.x3g
mr2x_bp5l_housingsurround.x3g
mr2x_left_gauge_holder_v2.x3g
mr2x_panel_holder.x3g
mr2x_simple_knob.x3g
```

while in a card reader on my desktop, I see:

```
/media/ryan/PRINT2$ ls -la
total 21188
drwxr-xr-x  2 ryan ryan    16384 Dec 31  1969 .
drwxr-x---+ 4 root root     4096 Feb 13 11:21 ..
-rw-r--r--  1 ryan ryan 13568878 Feb 10 14:54 MR2X_AAHolder3.x3g
-rw-r--r--  1 ryan ryan   369830 Feb  7 15:17 MR2X_BP5L-Housingbase.x3g
-rw-r--r--  1 ryan ryan   253465 Feb  7 15:15 MR2X_BP5L_housingsurround.x3g
-rw-r--r--  1 ryan ryan  1482695 Feb 12 11:26 MR2X_Left_Gauge_Holder_v2.x3g
-rw-r--r--  1 ryan ryan   554843 Feb  8 13:04 MR2X_panel_holder.x3g
-rw-r--r--  1 ryan ryan   385600 Feb 12 15:08 MR2X_simple_knob.x3g
-rw-r--r--  1 ryan ryan  4942921 Feb 13 10:25 MR2X_SolderingFingersHolder.x3g
```

DrLex0, 2018-06-21:
I should have seen this sooner, then I wouldn't have been juggling SD cards and checking their filesystems like crazy because it appeared something was corrupt. It seems indeed that certain filenames cause the printer to fail to see the files. The ‘logic’ behind this is very unclear and sometimes the same file will either appear or disappear depending on the presence of other files.
For instance, I've put two files inside a subdirectory of the SD card, called ‘SoccerThing-penta-CPR.x3g’ and ‘SoccerThing-hexa1-GLD.x3g’. The first was visible but the second wasn't. After renaming the latter file and adding some others, the first file was no longer visible.

LessPaul, 2018-07-06:
Short answer: Filenames are limited to 26 characters (30 including".xg3")
https://support.makerbot.com/troubleshooting/earlier-products/replicator-original/i-cant-find-my-file-on-the-sd-card_11732

DrLex0, 2018-07-07:
The files that were invisible in my case had names of only 25 characters. I have a suspicion that it may be a more general bug with the menu system in the latest build, but this is a pain to debug.

rpavlik, 2018-07-07:
Fwiw (I don't know how connected the menu system is to the UART connection)
- it's not just in the menu that these files don't show up, they also don't
show up when octoprint queries over USB for the SD file most (that's how I
could get copy-pastable stuff in the report above)

uoitalia, 2018-07-08:
May I ask two questions related to this:

1) How many files (and directories) are present in the directory were you can't see the file

2) Can you give a precise list of files present in a directory were this bug appears, it's ok for a screenshot or a txt list, at least to make it reproducible elsewere.

The 26 characters limit is correct, but I think too that there is something else, and I already have at least one idea...