jetty840/Sailfish-MightyBoardFirmware #199: Flashing bootloader to Mightyboard

nicoco007, 2018-04-13:
I've been trying to flash both the standard bootloader and the Optiboot build in this repository to a Mightyboard Rev G but they both make the board get stuck in some kind of boot loop: the led keeps flashing incessantly (for the normal ATmegaBOOT, it flashes about twice per second, while with Optiboot it flashes constantly). This is the command I've been using to flash:

```/Users/username/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -C /Users/username/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -p m1280 -F -P /dev/cu.usbmodem1451 -U flash:w:optiboot_atmega1280.hex -U lfuse:w:0xFF:m -U hfuse:w:0xDE:m -U efuse:w:0xF4:m -U lock:w:0x0F:m -c avrisp -e -b 19200```

I'm using an Arduino with ArduinoISP installed. The bootloader flashes without an issue, but it keeps resetting. I tried uploading firmware from ReplicatorG in this state and it's able to send a bit of data and then fails (TX/RX flashes quickly for about a second then stops). Any ideas why this isn't working?

tinder-danielnewman, 2018-04-13:
Contacting @Joachim if you have difficulty with optiboot.

As to changing the fuse bits for the ATmegaBOOT_168_atmega1280.hex bootloader, you should have left the fuse bits unchanged.  You should reset them to whatever they were before you started. (Sorry, off the top of my head I do not know the correct fuse bit settings on a Mightyboard Rev G board.)

nicoco007, 2018-04-13:
I just read them off another board and I get 0xDA for H (which is what I used when burning ATmegaBOOT_168_1280.hex), 0xF4 for E and 0xFF for L (didn't see anything for lock bits). I don't think that's the issue.

nicoco007, 2018-04-13:
Now that I think about it, I had set the extended fuse to 0xF5 at some point. Could that have bricked the board? Could mismatched lock bits also do this?

My only other hypothesis at this point is that the programmer isn't working properly.

nicoco007, 2018-04-14:
Alright, I just tried running ```/Users/username/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -C /Users/username/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -p m1280 -F -P /dev/cu.usbmodem1451 -U flash:w:ATmegaBOOT_168_atmega1280.hex -c avrisp -e -b 19200``` (no changes to fuses, just burn the bootloader) and it's doing the same thing. Any ideas?

(sorry about the spam)

tinder-danielnewman, 2018-04-14:
Sorry, no ideas.  As an aside, I've never had good luck using the `/dev/cu.usbmodemNNNN` device variant on a Mac.  But then, I also always use real AVR ISP programmers and avoid using an Arduino to do ISP programming. (They typically cannot burn more than 64K and the actual firmware is a few bytes shy of 124K.)

nicoco007, 2018-04-14:
Ah, okay, that might be the issue. I'll try on Linux first, and if that doesn't work I guess I'll have to invest in an actual ISP and go from there. Thanks for your help!

closed 2018-04-14 by nicoco007

sbts, 2018-06-01:
@nicoco007 would be good if you can report your results with trying Linux and / or a hardware ISP device