jetty840/Sailfish-MightyBoardFirmware #204: Thermal runaway protection?

rpavlik, 2018-10-09:
As this has been in various printing news recently, it would be nice to have this in the firmware. I didn't find anything with a quick search for runaway in the GitHub project, but it's possible I missed it.

I'd hope that at least the algorithm could be ported from Marlin since that seems to be the standard.

TehMillhouse, 2019-03-13:
grep for "HEATER_FAIL_". There are some basic sanity checks in place which you can test by, for instance, disconnecting the thermistor from your Hotend and turning on the heater cartridge. I don't know how exhaustive the checks are, but basic heater / thermistor failure seems to be caught.

uoitalia, 2019-03-14:
It only needs to add An abort print on the thermal shutdown case

Halbelunge, 2019-11-19:
I had the same question for my device with sailfish. So I checked the program code.
This can be found in the Heater.hh and Heater.cc:

enum HeaterFailMode{
	HEATER_FAIL_NONE = 0,
	HEATER_FAIL_NOT_PLUGGED_IN = 0x02,
	HEATER_FAIL_SOFTWARE_CUTOFF = 0x04,
	HEATER_FAIL_NOT_HEATING = 0x08,
	HEATER_FAIL_DROPPING_TEMP = 0x10,
	HEATER_FAIL_BAD_READS = 0x20
};

This is what is the interesting part: HEATER_FAIL_NOT_HEATING = 0x08,

Furthermore in the Heater.cc:

	  // check that the heater is heating up after target is set
	  if(!progressChecked){
	       if(heatProgressTimer.hasElapsed()){
		    if(current_temperature < (startTemp + HEAT_PROGRESS_THRESHOLD )){
			 value_fail_count++;

			 if (value_fail_count > SENSOR_MAX_BAD_READINGS) {
			      fail_mode = HEATER_FAIL_NOT_HEATING;
			      fail();
			      return;
			 }
		    }
		    else
			 progressChecked = true;
	       }

This seems to do exactly this check, what you mean with thermal runaway.

In my opinion, this can be closed.
Did you ever checked @rpavlik ?

SaschaKP, 2020-01-07:
What they mean is that even if that code is run, it doesn't prevent the printer from continuing the print, even if the heater is shut down, this can cause some hardware damage, as the filament is continuosly pushed in. My code option was already done in my fork, as this tree is INACTIVE and abandoned.

Halbelunge, 2020-06-04:
Ahh ok, I see.
But hey, at least it will not burn your house down, because of an open loop in the thermal regulation.