jetty840/Sailfish-MightyBoardFirmware #210: Mightyboard rev E thermistor on PK5 and PK6

jgrabitzky, 2020-04-22:
My Board has Thermistor on PK5 and PK6. No compiled Sailfish Version can read the temperatur. 
I need a compiled Version or help please.
I have a board with atmel2560 prozessor.

[image]

SaschaKP, 2020-07-03:
you have to change the pins inside the sources, consider switching to MARLIN, since version 2.0.x bugfix WORKS with replicator 1&2 clones, you just have to tweak pins positions.
[image]

caall99, 2020-07-03:
Is the Marlin firmware better than the Sailfish firmware now?

SaschaKP, 2020-07-03:
in terms of configurability and easy mods yes, and since I've maintained my fork of sailfish, for changes relative to unfinished systems or simple mods, I know of what I'm talking about, if you're in search of the 150mm/s, that's impossible to achieve without a really strong hardware and only for certain items, for the rest and for an unattended printer I can say absolutely yes.

Just consider the thermistor problem above, you have to do changes directly in the source to make it work in sailfish (and also figure out how to do it), while in marlin you have to do only a pin assignment in mightyboardreve pin config file and specify the type of temperature sensor in the config.h file.

Not only that, I consider the use of the autoleveling probe a must, and since I have one angle of the bed that has a 0.08mm bend, non fixable with screws, the only solution would be to rebuy some hardware, that for that ancient printer is a no-no.

Better? absolutely yes, faster? possibly not.

caall99, 2020-07-03:
Thank you for the information. Where might I find this marlin firmware
compatible with my Qidi Tech 1?

SaschaKP, 2020-07-05:
you'll have to compile it yourself, it's a replicator 1 clone anyway, as I've seen in the web by researching your printer type.

evilexecutable, 2020-12-02:
Hate to grave dig, but I'm in the same boat. In pins_MIGHTBOARD_REVE.h, am I to change teh THERMO_CS1_PIN to point to PK6?