jetty840/Sailfish-MightyBoardFirmware #214: Scons cannot compile

TheG0bbler, 2021-06-14:
Error presents on Python 3.9.5 and 3.9.2

Input: 
```
>scons platform=ff_creator-2560
scons: Reading SConscript files ..
```

Error:
```
C:\Users\USER\Desktop\Sailfish-MightyBoardFirmware-master\firmware>scons platform=ff_creator-2560
scons: Reading SConscript files ...
TypeError: a bytes-like object is required, not 'str':
  File "C:\Users\USER\Desktop\Sailfish-MightyBoardFirmware-master\firmware\SConstruct", line 6:
    SConscript(['src/SConscript.mightyboard'], variant_dir='build/'+platform)
  File "c:\users\USER\appdata\local\programs\python\python39\lib\site-packages\SCons\Script\SConscript.py", line 654:
    return method(*args, **kw)
  File "c:\users\USER\appdata\local\programs\python\python39\lib\site-packages\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "c:\users\USER\appdata\local\programs\python\python39\lib\site-packages\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\USER\Desktop\Sailfish-MightyBoardFirmware-master\firmware\build\ff_creator-2560\SConscript.mightyboard", line 138:
    vstr = vstr.split('\n')[0]
```

jeffsauer, 2021-06-18:
I think Sailfish requires you to use a Python2 environment.

I'm on Arch linux, so I installed the Python2 version of scons

yay -S python2-scons

which will then give you a /usr/bin/scons2 command binary.  To be safe, in case the build scripts called scons and not scons2, I also created a symlink:

sudo ln -s /usr/bin/scons2 /usr/bin/scons

and I was able to then compile cleanly.

jeffsauer, 2021-06-18:
BTW, I have a FFCP with a mightyboard 2560 Rev E... not sure if I should use platform ff_creator-2560 or mighty_one-2560 ... the .hex images they each build are indeed different.